gpiozero/gpiozero #709: Should rpio be dropped from the list of backends that are tried by default.

plugwash, 2019-01-17:
rpio seems rather bitrotten. It seems to be lacking any form of pi detection code detecting neither whether it is on a Pi at all or what model of Pi is in use. It just assumes a pi1. Other than an update to the readme it doesn't seem to have had any activity since 2015.

should it be removed from the list of pin factories to try by default?

bennuttall, 2019-01-17:
I have never used it myself, but @waveform80 added it for its hardware PWM support (which none of the others have). Unfortunately it seems to have been abandoned, and therefore does not support Pi2 and up.

I assume it works on a Pi Zero - does anyone know?

I don't see any harm in it being there, if it at least has some small purpose.

bennuttall, 2019-01-17:
I don't see a compelling reason to remove it from the list if it could be picked up there. Why do you think so?

lurch, 2019-01-17:
Seems like there are a lot of people trying to get it functional again, but I dunno if any forward steps are actually being taken? https://github.com/metachris/RPIO/issues/104
ping @metachris 

Although I agree that there doesn't seem to be much reason for removing it from GpioZero - if somebody has gone to the effort of installing rpio on their Pi they presumably have a good reason for doing so?

bennuttall, 2019-01-19:
I tried it today and it doesn't even work on a modern OS image on a Pi 1.

Still, there's no harm it being there - and if someone picks up the project and gets it working again, then it'll be nice to have.

closed 2019-01-23 by bennuttall