gpiozero/gpiozero #712: pigpio pwm does not support float frequencies

bennuttall, 2019-01-20:
Operating system: Raspbian Stretch
Pi model: e.g. Pi 3 Model B+
GPIO Zero version: e.g. 1.4.1
Pin factory used: e.g. pigpio

RPiGPIOPin will happily take float values as its PWM device frequency, but pigpio falls over if you try to send a float. The following code will work for RPiGPIOPin but not for PiGPIOPin:

```python
>>> pwm = PWMOutputDevice(20)
>>> pwm.frequency = 500.0
```

The following error occurs (truncated stack trace):

```python
/usr/lib/python3/dist-packages/pigpio.py in _pigpio_command(sl, cmd, p1, p2, rl)
    981    """
    982    sl.l.acquire()
--> 983    sl.s.send(struct.pack('IIII', cmd, p1, p2, 0))
    984    dummy, res = struct.unpack('12sI', sl.s.recv(_SOCK_CMD_LEN))
    985    if rl: sl.l.release()

error: required argument is not an integer
```

Also the pigpio connection seems to hang after this, so you can't carry in after an error like this in the shell. You can't even exit the shell without forcing it.

As discussed with @waveform80 we need to address this in pins API in v1.6.

ForToffee, 2019-06-17:
Just tripped over this using gpiozero on x86 raspbian to a pizero as gpio extender.

I've put in a work around to int() the value before it is passed to the factory in _set_frequency in pigpio.py.

Any idea when this will be fixed?

bennuttall, 2019-06-17:
> I've put in a work around to int() the value before it is passed to the factory in _set_frequency in pigpio.py.

Yeah I tried that but it's more complicated. I'll take a look when I get chance but I think it's probably one for @waveform80.

ForToffee, 2019-06-18:
From what I experienced last night the issue appears to be self.factory.connection.set_PWM_frequency() on lines 300 and 306.  Wrapping value on these two lines in int() fixed it.  What's the bit I'm missing if this isn't the correct way of resolving this?

It's a pain as I'm setting up a session using laptops and PiZeros in extender mode which relies on pigpio

bennuttall, 2019-06-18:
Did it fix it in a way that made TonalBuzzer work properly, or just make the error go away? I seem to recall only managing the latter

ForToffee, 2019-06-18:
Yes, TonalBuzzer does work (i.e. it plays a tone, rather than error out).  I've (hopefully) created a PR for you.

PR #933 (2021-03-12): Fix for #712
convert value being passed to set_PWM_frequency() to int to ensure float value is not passed to pigpio daemon

fixes #712
closed 2021-03-12 by waveform80