gpiozero/gpiozero #714: Disk usage internal device

bennuttall, 2019-01-22:
`df -h` as an internal device.

jee1mr, 2019-01-23:
@bennuttall I can take up this too?

bennuttall, 2019-01-23:
Go for it :+1:

Don't PR it just yet - I'm adding tests and changing things in that file. If you just create a new class you can paste in later on that should be great.

It should take an argument which defaults to `/` for which drive it's looking at.

I was going to build on this: https://github.com/bennuttall/piwheels/blob/227648a1e8296f676da9c93816ba9355d12e8b75/piwheels/tools.py#L37:L40

jee1mr, 2019-01-23:
```python
class DiskUsage(InternalDevice):
    """
    Extends :class:`InternalDevice` to provide a device which is active when
    the disk free space exceeds the *threshold* value.

    :param str filesystem:
        The filesystem for which the disk usage needs to be computed. This
        defaults to `/`, which is the root filesystem. 

    :param float threshold:
        The disk usage percentage above which the device will be considered "active".
        This defaults to 90.0.
    """
    def __init__(self, filesystem='/', threshold=90.0):
        super(DiskUsage, self).__init__()
        self.filesystem = filesystem
        self.threshold = threshold
        self._fire_events()

    def __repr__(self):
        return '<gpiozero.DiskUsage usage=%.2f>' % self.usage

    @property
    def usage(self):
        """
        Returns the current disk usage in percentage.
        """
        df = subprocess.Popen(['df', '-h', self.filesystem], stdout=subprocess.PIPE)
        output = df.communicate()[0].split()
        return float(output[11].decode('UTF-8').rstrip('%'))

    @property
    def is_active(self):
        """
        Returns ``True`` when the Disk :attr:`usage` exceeds the
        :attr:`threshold`.
        """
        return self.usage > self.threshold
```

@bennuttall Does it look good?

lurch, 2019-01-24:
I think the docstring needs changing from "a device which is active when the disk free space exceeds the *threshold* value" to "a device which is active when the disk space used exceeds the *threshold* value" ?

Might be worth splitting on newlines _before_ splitting on whitespace, in case the column headers have a different number of spaces in different languages? (although a quick skim of `df --help` suggests `df --output=pcent <filesystem>` might be easier? :wink: )

Perhaps it's also worth providing a `value` property which is simply the usage % re-scaled from 0-100 to 0-1? (which then means it would plug in nicely with sourcetools, e.g. have a LED bargraph displaying how full your disk is :slightly_smiling_face: )

bennuttall, 2019-01-24:
Thanks both - @jee1mr can you PR this with @lurch's changes?

jee1mr, 2019-01-24:
@lurch Thanks for the suggestions :)
@bennuttall Will do.

commit 1508d9c7 (2019-01-25): Add DiskUsage (#720), close #714

Add DiskUsage internal device, close #714