gpiozero/gpiozero #715: Check for ZeroDivisionError on init

bennuttall, 2019-01-22:
```python
In [1]: from gpiozero import *                                                                       

In [2]: cpu = CPUTemperature(min_temp=100)                                                           

In [3]: cpu.value                                                                                    
---------------------------------------------------------------------------
ZeroDivisionError                         Traceback (most recent call last)
<ipython-input-3-4e90eb16ea73> in <module>
----> 1 cpu.value

/usr/lib/python3/dist-packages/gpiozero/other_devices.py in value(self)
    144         """
    145         temp_range = self.max_temp - self.min_temp
--> 146         return (self.temperature - self.min_temp) / temp_range
    147 
    148     @property

ZeroDivisionError: float division by zero
```

jee1mr, 2019-01-23:
Hi @bennuttall. I've been using gpiozero for a while now and I would love to contribute back.  This seems like a good one to begin with. Shall I pick it up if you haven't started on it already?

jee1mr, 2019-01-23:
`diff --git a/gpiozero/other_devices.py b/gpiozero/other_devices.py
index 219e1a9..cd9208a 100644
--- a/gpiozero/other_devices.py
+++ b/gpiozero/other_devices.py
@@ -118,8 +118,11 @@ class CPUTemperature(InternalDevice):
             min_temp=0.0, max_temp=100.0, threshold=80.0):
         self.sensor_file = sensor_file
         super(CPUTemperature, self).__init__()
+        if min_temp >= max_temp:
+            raise ValueError('min_temp must be less than max_temp')
         self.min_temp = min_temp
         self.max_temp = max_temp
+        self._temp_range = self.max_temp - self.min_temp
         self.threshold = threshold
         self._fire_events()

@@ -142,8 +145,7 @@ class CPUTemperature(InternalDevice):
         *max_temp* value). These default to 0.0 and 100.0 respectively, hence
         :attr:`value` is :attr:`temperature` divided by 100 by default.
         """
-        temp_range = self.max_temp - self.min_temp
-        return (self.temperature - self.min_temp) / temp_range
+        return (self.temperature - self.min_temp) / self._temp_range

     @property
     def is_active(self):`

Would this do?

jee1mr, 2019-01-23:
Opened a PR #716