gpiozero/gpiozero #730: Ensure every class's repr is tested

bennuttall, 2019-02-10:
It's possible for all tests to be passing but a device's repr fail, as I just found with `TonalBuzzer`.

I added a call to `repr(tb)` at the top of the first test for `TonalBuzzer` and since calling it raised an exception.

I know some tests actually inspect the contents of repr. Do we think this is necessary for all classes? Maybe. I think there should be at least a "test" that repr doesn't fall over.

@waveform80 @lurch thoughts?

lurch, 2019-02-10:
I agree with @bennuttall - every class's `repr` should at least be callable (at any time) without causing an Exception, but I don't think there's any need to explicitly test the contents of each repr, since it's an "internal detail" and not part of the GpioZero API. All IMHO.

bennuttall, 2019-02-10:
There's no harm in testing the implementation e.g. tonal buzzer's repr should now be `<TonalBuzzer playing tone A4>` but as I say, we should *at least* test that repr actually works.

lurch, 2019-02-10:
> I don't think there's any need to explicitly test the contents of each repr

...although I guess that you could at least test that `repr()` returns a `str` with length > 1, and if you wanted to get fancy you could check it starts with `<gpiozero.`

bennuttall, 2019-02-10:
That makes sense

PR #732 (2019-02-10): Add repr tests, close #730
Add repr tests and make internal device repr uniform (and use super when it fails)

Fixes #730
commit 4e4551a5 (2019-02-10): Merge pull request #732 from RPi-Distro/repr-tests

Add repr tests, close #730
closed 2019-02-10 by bennuttall