gpiozero/gpiozero #731: Build docs warnings

lurch, 2019-02-10:
I just followed the instructions [here](https://gpiozero.readthedocs.io/en/master/development.html) on my Ubuntu 16.04 PC, and when I run `make doc` I get a whole load of scary-looking warnings displayed in my terminal (in red), along the lines of:
```
Running Sphinx v1.8.4
making output directory...
loading intersphinx inventory from https://docs.python.org/3.5/objects.inv...
loading intersphinx inventory from https://picamera.readthedocs.io/en/stable/objects.inv...
WARNING: failed to reach any of the inventories with the following issues:
WARNING: intersphinx inventory 'https://picamera.readthedocs.io/en/stable/objects.inv' not fetchable due to <class 'requests.exceptions.HTTPError'>: ('intersphinx inventory %r not fetchable due to %s: %s', 'https://picamera.readthedocs.io/en/stable/objects.inv', <class 'requests.exceptions.HTTPError'>, HTTPError(...))
loading intersphinx inventory from https://colorzero.readthedocs.io/en/stable/objects.inv...
WARNING: failed to reach any of the inventories with the following issues:
WARNING: intersphinx inventory 'https://colorzero.readthedocs.io/en/stable/objects.inv' not fetchable due to <class 'requests.exceptions.HTTPError'>: ('intersphinx inventory %r not fetchable due to %s: %s', 'https://colorzero.readthedocs.io/en/stable/objects.inv', <class 'requests.exceptions.HTTPError'>, HTTPError(...))
```

Am I missing some dependency, or are these harmless warnings that can be ignored?

bennuttall, 2019-02-10:
@waveform80 has done considerable work in the [docs-tidy-up](https://github.com/RPi-Distro/python-gpiozero/tree/docs-tidy-up) branch recently. I think most/all such warnings  will have gone away now.

lurch, 2019-02-10:
Yup, switching to the `docs-tidy-up` branch gets rid of the intersphinx warnings :+1: Sorry I didn't try that earlier!
But there's now a red warning that `copying static files... WARNING: html_static_path entry '/home/qpc/python-gpiozero/docs/_static' does not exist` ?

closed 2019-02-10 by bennuttall

lurch, 2019-02-10:
Does this being closed imply that the warning is indeed harmless? :)

bennuttall, 2019-02-12:
I believe so

waveform80, 2019-02-13:
It is harmless, but I fixed it anyway in a commit just before release (just added a placeholder file as git won't track empty dirs).