gpiozero/gpiozero #738: Request for better MockPin docs

lurch, 2019-02-11:
I was just trying to locally construct a PWMLED object on a MockPWMPin in a Python3 prompt on my Ubuntu PC, and with all the pin-factory, pin-spec, etc. gubbins I was unable to work out how to perform this (seemingly simple) task :frowning_face: Everything I tried (e.g. `pled = PWMLED(MockPWMPin(7))`) just threw an exception at me.

Also, it might be useful if https://gpiozero.readthedocs.io/en/master/development.html linked to https://gpiozero.readthedocs.io/en/master/api_pins.html#mock-pins ?

bennuttall, 2019-02-11:
Absolutely agree - and I went on the same learning curve you did a couple of weeks ago when my old method of using mock PWM pins failed due to recent changes. I'd welcome a docs page dedicated to mock, or it could just go in development.

In case you didn't figure it out in the end:

```python
>>> from gpiozero import *
>>> from gpiozero.pins.mock import MockFactory, MockPWMPin
>>> Device.pin_factory = MockFactory(pin_class=MockPWMPin)
>>> led = PWMLED(2)
```

:+1:

lurch, 2019-02-11:
Aha, thanks Ben! :+1: 
Is it no longer possible to have a mix of MockPins and MockPWMPins? Or do I need to do that by constantly switching the `Device.pin_factory` ? :confused: 

Also, why does the MockPWMPin constructor take a `factory` parameter? Won't it always be the MockFactory?

bennuttall, 2019-02-11:
> Is it no longer possible to have a mix of MockPins and MockPWMPins? Or do I need to do that by constantly switching the `Device.pin_factory` ?

Don't think that's been possible for ages - not since we added the `pin_factory` keyword. But that way is the easiest way - remember `Device.pin_factory` is just the default.

```python
In [1]: from gpiozero import *                                                                       
In [2]: from gpiozero.pins.mock import MockFactory, MockPWMPin                                       
In [3]: led = LED(2)                                                                                 
In [4]: pwmfactory = MockFactory(pin_class=MockPWMPin)                                               
In [5]: pwmled = PWMLED(3, pin_factory=pwmfactory)
```

lurch, 2019-02-11:
Aha, thanks again @bennuttall :+1: 
So if I want to be explicit I can do:
```python
from gpiozero import LED, PWMLED
from gpiozero.pins.mock import MockFactory, MockPin, MockPWMPin
mockfactory = MockFactory(pin_class=MockPin)
mockpwmfactory = MockFactory(pin_class=MockPWMPin)
led = LED(2, pin_factory=mockfactory)
pwmled = PWMLED(3, pin_factory=mockpwmfactory)
```
And I can then access the MockPin and MockPWMPin objects with `led.pin` and `pwmled.pin`. Cheers!
(but this clearly needs better docs - I guess this has all gone through so many changes over the years that my brain can't keep up :laughing: )

bennuttall, 2019-02-11:
:+1: that's a good method, yes.

And yes, we should add to the docs. I said to Dave earlier that while I know how to play with leds and buttons in mock mode I couldn't simulate a distance sensor - he said look in the test suite! True, but we should document it too.

bennuttall, 2019-02-14:
I wonder if it would be possible to have an environment variable to set the pin class (purely for MockPWMPin I think). It would be very handy to be able to launch IPython and be able to use mock pins without messing around (or, inevitably, trying to create a device before realising you need to set the pin class).

@waveform80 is that too much hassle?

lurch, 2019-02-15:
Something like:
```
GPIOZERO_PIN_FACTORY=mock GPIOZERO_MOCK_PIN_CLASS=MockPWMPin ipython3
```
?

bennuttall, 2019-02-15:
Yep - just not certain it's practical

lurch, 2019-02-15:
And on a related note, it would also be nice if https://gpiozero.readthedocs.io/en/master/development.html explained how to run just a single test (or a subset of tests) rather than the whole test-suite :)  (for when you're doing rapid iteration)

bennuttall, 2019-02-15:
Good suggestion. Are you also asking?

lurch, 2019-02-15:
Yup ;)

bennuttall, 2019-02-15:
```python
pytest tests/test_boards.py  # run all tests in one file 
pytest -k jamhat  # (search) run all tests with "jamhat" in the test name
```

I tend to use `-v` for verbose output (useful when looking at a diff), so usually `pytest -v` or e.g. `pytest `-vk jamhat`

lurch, 2019-02-15:
...and it's probably also worth documenting that for MockPins, changing the pull-state _always_ also changes the pin's state, i.e.
```python
>>> from gpiozero import *
>>> from gpiozero.pins.mock import *
>>> Device.pin_factory = MockFactory()
>>> pin = Device.pin_factory.pin(5)
>>> pin.pull = 'down'
>>> pin.drive_high()
>>> pin.state
True
>>> pin.pull = 'down'
>>> pin.state
False
```

It's just taken me ages to track down that this is why I was seeing such odd behaviour in one of my failing tests :wink: 

I wonder if it'd make more sense for MockPin's `_set_pull` to do:
```python
    def _set_pull(self, value):
        if self.function != 'input':
            raise PinFixedPull('cannot set pull on non-input pin %r' % self)
        if value != 'up' and self.factory.pi_info.pulled_up(repr(self)):
            raise PinFixedPull('%r has a physical pull-up resistor' % self)
        if value not in ('floating', 'up', 'down'):
            raise PinInvalidPull('pull must be floating, up, or down')
        old_pull = self._pull
        self._pull = value
        if old_pull == 'floating' or old_pull != self._pull:
            if value == 'up':
                self.drive_high()
            elif value == 'down':
                self.drive_low()
```
....although I just tried that and it caused `test_output_initial_values` to fail on my Ubuntu PC which is kinda odd, and I don't have time to unravel that loose thread any further right now :laughing:

lurch, 2019-02-15:
> I wonder if it would be possible to have an environment variable to set the pin class (purely for MockPWMPin I think). It would be very handy to be able to launch IPython and be able to use mock [...]

https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/mock.py#L469
:joy: :rofl:

lurch, 2019-02-15:
Been doing some debugging of my problem outlined above, namely that:
```python
    pin = mock_factory.pin(2)
    with OutputDevice(2, initial_value=False) as device:
        assert not pin.state
        state = pin.state
    with OutputDevice(2, initial_value=None) as device:
        pin.debug(3)
        assert state == pin.state
```
fails, because on the last line `pin.state` is True when we expect it to be False.

Did a bit of thread-unravelling, and it seems to be failing because the first `with` block means that the  device gets closed when we exit the with-block, and MockPin's `close` does:
```python
    def close(self):
        self.when_changed = None
        self.function = 'input'
```
and MockPin's function-setter does:
```python
    def _set_function(self, value):
        if value not in ('input', 'output'):
            raise PinInvalidFunction('function must be input or output')
        self._function = value
        if value == 'input':
            # Drive the input to the pull
            self._set_pull(self._get_pull())
```
and MockPin's pull-setter does:
```python
    def _set_pull(self, value):
        if self.function != 'input':
            raise PinFixedPull('cannot set pull on non-input pin %r' % self)
        if value != 'up' and self.factory.pi_info.pulled_up(repr(self)):
            raise PinFixedPull('%r has a physical pull-up resistor' % self)
        if value not in ('floating', 'up', 'down'):
            raise PinInvalidPull('pull must be floating, up, or down')
        old_pull = self._pull
        self._pull = value
        if value == 'up':
            self.drive_high()
        elif value == 'down':
            self.drive_low()
```
with drive_high then (eventually) setting the MockPin's `._state` to True.

So, bringing that all together:
 1. We create a OutputDevice passing initial_value=False, which sets the MockPin's `._state` to False
 2. When the OutputDevice exits the with-block, it then gets closed
 3. OutputDevice doesn't have a close method, so the parent GPIODevice.close() gets called
 4. GPIODevice's close method closes the MockPin
 5. MockPin's close method sets the pin's function to input
 6. MockPin's function-setter does `self._set_pull(self._get_pull())`
 7. The default pull for GPIO2 is `up` https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/data.py#L284
 8. Because we're setting the pull back to 'up', the pin gets driven high
 9. And therefore the MockPin's `._state` gets set to True
 10. Which means when we re-open GPIO2 as an OutputDevice passing initial_value=None, we get the pin's state as True (because that's what it was set to during close), and not the last-output value of False that we might expect!

PHEW, that was hard work! :sweat_smile: 

I'm away from home ATM so don't have a real Pi I can test with to see if the real pins behave in the same way. But in the meantime if I change `test_output_initial_values` to use GPIO4 instead of GPIO2 then it goes back to passing again :slightly_smiling_face: 
I'm fairly sure I once suggested before that all the tests and examples should be updated to use GPIO pins without pull-ups, precisely to avoid these kind of unexpected side-effects... :stuck_out_tongue_winking_eye:

lurch, 2019-02-17:
...or perhaps a better approach is to "not care" what closing the device / pin does (see also #707 ), and simply modify the `test_output_initial_values` to explicitly test only what the init-method does (i.e. set the pin to a known output state and check that `initial_value=None` doesn't modify that output state)
```patch
--- a/tests/test_outputs.py
+++ b/tests/test_outputs.py
@@ -57,9 +57,16 @@ def test_output_initial_values(mock_factory, pwm):
         assert repr(device).startswith('<gpiozero.OutputDevice object')
         assert pin.function == 'output'
         assert not pin.state
+        state = pin.state
+    pin.function = 'output'
+    pin.state = state
+    with OutputDevice(2, initial_value=None) as device:
+        assert state == pin.state
     with OutputDevice(2, initial_value=True) as device:
         assert pin.state
         state = pin.state
+    pin.function = 'output'
+    pin.state = state
     with OutputDevice(2, initial_value=None) as device:
         assert state == pin.state
 
```
?

bennuttall, 2019-09-22:
@lurch see an update here : https://github.com/gpiozero/gpiozero/issues/738#issuecomment-462310186

thought you'd find it useful :)

lurch, 2019-09-23:
...which I pointed to here https://github.com/gpiozero/gpiozero/issues/738#issuecomment-464009339 :wink:

commit 7acfdb24 (2019-09-23): Extend mock pin documentation, close #738
closed 2019-09-23 by bennuttall