gpiozero/gpiozero #746: Class init documentation inconsistencies

lurch, 2019-02-13:
While looking through the docs I just happened to notice that https://gpiozero.readthedocs.io/en/master/api_input.html#smoothedinputdevice says:
```
class gpiozero.SmoothedInputDevice(pin, *, pull_up=False, active_state=None, threshold=0.5, queue_len=5, sample_wait=0.0, partial=False, pin_factory=None)
```
but then it elaborates the list of parameters below that as:

- pin
- pull_up
- active_state
- threshold
- queue_len
- sample_wait
- partial
- average
- ignore
- pin_factory

Note that `average` and `ignore` are in the second list but not in the first list, eek! :confounded: 
I'm not au-fait with how the docs are generated so I didn't realise it was possible to get inconsistencies like this. I don't have time to check right now, but maybe it affects other classes' docs too?

waveform80, 2019-02-21:
Yup, entirely possible to get inconsistencies in this area. There's two reasons this can happen:

* The list defining what the parameters mean (the one with the extra entries) is entirely manually written; it's not generated at all.

* The list defining the call prototype (`class SmoothedInputDevice(pin, *, blah, blah, blah, ...)`) *should* be automatically generated, and in most sphinx documentation is. However, in GPIO Zero we've done things a little differently. Because we're (currently) maintaining compatibility with Python 2.7 we don't have the option of *actually* declaring keyword-only args (i.e. everything after the "*" parameter) because that's not valid syntax in Python 2.x. Instead those parameters are optional positional/keyword parameters. We *intend* them to be keyword-only so we override the docs (in the `api_*.rst` files) to list them as such in the vague hope that people will take the documentation as the "contract" between our library and them, and will use them as keyword-only args even though they're aren't (currently). Once we (eventually) manage to ditch Python 2.x compatibility we can remove these overrides and this part at least should become consistent with what's in the code.

waveform80, 2021-03-19:
We've now ditched py2.x and I've just spent an evening excising all our overrides from the master branch of the docs so this *should* be fixed :)

closed 2021-03-19 by waveform80