gpiozero/gpiozero #747: New Feature Request - GPIO and Desktop Via Ethernet

JonnyAlpha, 2019-02-13:
Operating system: Raspbian Stretch
Pi model: Pi 3 Model B (with backwards compatibility where able)

**Background:**
I am a Primary School computing teacher / Raspberry Pi Cert Ed and due to space and time limitations have only been able to use Raspberry Pis on a very limited scale in an after school club. 
I understand that a user can attach a Pi Zero and with Rasbian Desktop running via DVD or installed on the Laptop a user can access the GPIO pins. I have asked our school IT contractor if we have the ability to boot from CD on our Windows 7 Client devices, which would then allow us to use PiZeros (If we had them). However I would like to be able to take this further as follows. What I would like to see is the ability to simply plug a Raspberry Pi 3 into a Laptop via Ethernet and run VNC Server on the Host PC and have realtime access to the Raspberry Pi Desktop and GPIO and other interfaces.

**New Feature Request:**
I would like to see the ability to connect a Raspberry Pi into a host device via an Ethernet patch cable power up the Pi either using a separate power supply or possibly the host PC's USB (power limitations) and run VNC Viewer on the Host device and have live access to the connected RPi. 

VNC Server would need to run at boot and the Pi would need a static IP Address. I am not an expert in many things and understand that the static IP may conflict with the host device unless there is a way around this. 

Whilst I understand this is not a GPIO Zero issue but I have asked elsewhere and many users don't see this as an issue because (as experts) they know the workarounds. 
This feature would break down the barriers that current exist in education, which in many schools is the lack of a dedicated IT suite or technical experts that can use Putty and understand how to remote access. To add this functionality would allow 1000s of more children (and adults) access this fantastic device.

lurch, 2019-02-13:
Is https://www.raspberrypi.org/documentation/remote-access/ what you're looking for? :smiley: 

Or perhaps you're thinking more along the lines of https://www.raspberrypi.org/blog/piserver/ ?

lurch, 2019-02-13:
> VNC Server would need to run at boot and the Pi would need a static IP Address. I am not an expert in many things and understand that the static IP may conflict with the host device unless there is a way around this.

Or perhaps https://www.pibakery.org/ is what you need? :man_shrugging: 

Sorry for bombarding you with so many options, but as you can see the Pi is so flexible that there's many different ways of doing things.

closed 2019-09-20 by bennuttall