gpiozero/gpiozero #748: Why doesn't pinout -x open in chromium?

bennuttall, 2019-02-13:
The command `pinout -x` or `pinout --xyz` opens [pinout.xyz](https://pinout.xyz) in a browser, but for some reason not Raspbian's default one, Chromium.

This is not a gpiozero issue, as all we're doing is using `webbrowser.open()` - the same as `import antigravity`, but it would be nice if we could get to the bottom of this and suggest a change to the config Raspbian.

Here are the "browser" related alternates in Raspbian:

```
pi@raspberrypi:~$ update-alternatives --get-selections | grep browser
gnome-www-browser              auto     /usr/bin/chromium-browser
infobrowser                    auto     /usr/bin/info
netsurf-browser                auto     /usr/bin/netsurf-gtk
x-www-browser                  auto     /usr/bin/chromium-browser
```

I'm not sure which one is being used - it's not netsurf, it's called "Web". `/usr/bin/info` doesn't seem to contain anything about a browser.

`pinout -x` works fine on my Ubuntu PC - and opens in my default browser.

lurch, 2019-02-13:
This might be something @spl237 or @XECDesign can help with?

XECDesign, 2019-02-13:
Doesn't answer the question, but Web is what epiphany has been renamed to. Shouldn't be hard to figure out why it's being launched. We can take a look tomorrow.

spl237, 2019-02-14:
It's a problem with the Python webbrowser module; when it claims to open "the default browser", what it actually means is "the default browser, as long as it is in the list of candidate browsers hard-coded in the webbrowser module", and Chromium isn't in that list.

For reference, the list in question is on line 521 of webbrowser.py, and includes, in order, firefox, firebird, seamonkey, mozilla, netscape, Opera and IE - no Chromium. (Containing netscape as it does, this list was so clearly ahead of its time that it has never been thought to be necessary to update it... ;) )

To get your page to open in Chromium, you need to explicitly create a Chromium browser object to override the default list - replace

webbrowser.open ('https://pinout.xyz')

with

mybrowser = webbrowser.get ('chromium-browser')
mybrowser.open ('https://pinout.xyz')

spl237, 2019-02-14:
Sorry - didn't mean to close that!

bennuttall, 2019-02-14:
@spl237 I could do that but I'd have to add logic for "if this is a Pi" and "if they have chromium" (there's a chance they've removed it).

Hypothetically*, how much do you think would need to be changed in webbrowser for it to pick up the default browser properly?

`*` maybe not hypothetically

lurch, 2019-02-14:
Oh, that's interesting! But if we're overriding the default functionality of the webbrowser module, surely it'd be better to use the default browser as identified by the `update-alternatives` that Ben mentioned above, than to hardcode chromium-browser?

XECDesign, 2019-02-14:
I haven't looked into it, but it seems like it's trying to use xdg-open first, which should work. Is it failing to use xdg-open or is xdg-open opening the wrong browser?

spl237, 2019-02-14:
It looks to me as if it is failing to use xdg-open, as xdg-open falls over with an error on Buster, and I don't see that error when run from webbrowser. I'll check...

spl237, 2019-02-14:
On digging about, it turns out that webbrowser is using a rather unorthodox means of determining the default web browser. Instead of looking at the Debian alternatives system, it looks at the default application assigned to a rather obscure MIME type - i.e. rather than something sensible like text/html, it looks at what is assigned to handle x-scheme-handler/http. We don't define a default handler for this at present, so it looks at all the installed applications which say they can handle this type, and chooses the one which is last alphabetically. As "org.gnome.epiphany" is further down the alphabet than "chromium", it wins.

I've added chromium as an explicit default handler for that type, and pinout now opens in chromium. We'll push that into a raspberrypi-ui-mods release at some point.

bennuttall, 2019-02-14:
That's awesome, thanks Simon.

closed 2019-05-14 by bennuttall