gpiozero/gpiozero #749: PWM frequency at low duty cycle

mod38, 2019-02-14:
When driving a small cheap dc motor with PWM, the PWM frequency is unimportant at higher (>50%) duty cycle as all that is required is to control the average power. At low duty cycles however this does not provide enough torque to keep the motor turning. A frequency of around 10Hz provides a series of individual pulses that will keep a motor turning at surprisingly low speeds. It would be good therefore if there were a simple way to control the PWM frequency in GPIOZero motor commands.

lurch, 2019-02-14:
I know very little about electronics, but would using a smoothing capacitor (I have no idea what value might be appropriate) be a simple hardware workaround here?

mod38, 2019-02-14:
Thanks for responding Andrew but the problem is the other way around. At 100Hz with a 10% duty cycle the power at the motor is not very different from just feeding it 10% of the supply voltage, i.e. it is smoothed by the capacitance and reluctance in the system. At 10Hz however the motor will be getting individual pulses of the full voltage spaced 100ms apart. This gives it the kick it needs to keep turning. It's even more effective if the supply is higher than the continuous rating for the motor.

lurch, 2019-02-15:
Thanks for the explanation :+1:  Going in the other direction, would a 10 Hz PWM frequency not drive the motor successfully for duty cycles above 50% ? (again, please excuse my ignorance!)

> It's even more effective if the supply is higher than the continuous rating for the motor.

Does that imply that whatever class is used for this, it should avoid setting the duty cycle all the way up to 100%?

Do you have some kind of formula for how the PWM frequency should be altered as the duty cycle changes? Or is it something that needs to be determined by trial & error for each style of motor being used?

mod38, 2019-02-15:
Certainly 10Hz will work fine for motors at any duty cycle. The choice of 100Hz or more seems to be to  allow flicker-free dimming of LEDs. I have looked at the docs and in Output Devices, base classes, there is PWMOutputDevice which allows you to set the frequency - defaulted to 100Hz. This function obviously does not have the motor-specific features of the motor function such as forward/reverse.
The issue of not setting the power too high is really a hardware issue for the experimenter. The motors spin fairly wildly if you over power them! Since the point is to use them at low speed the issue doesn't really arise.

lurch, 2019-02-15:
Ah, well if 10Hz will also work fine for higher duty-cycles too, then I guess you might be able to do something like:
```python
motor = Motor(17, 18)
motor.forward_device.frequency = 10
motor.backward_device.frequency = 10

motor.forward(0.1)
```
?

lurch, 2019-02-15:
@bennuttall I guess this is another example like #751 where you want to pass additional init-params to a CompositeDevice's sub-devices :smiley:

mod38, 2019-02-15:
@lurch Am I right in thinking this is something you are suggesting could be added into the motor class? I have looked through the source for that and I see that it calls PWMOutputDevice if pwm is True so passing a frequency to to that class would indeed solve the problem. 
Separately I am trying to find / remember code I wrote where I used two pins to set the direction of the motor but applied pwm to an enable pin to control speed.

lurch, 2019-02-16:
> Separately I am trying to find / remember code I wrote where I used two pins to set the direction of the motor but applied pwm to an enable pin to control speed.

https://gpiozero.readthedocs.io/en/stable/api_output.html#phaseenablemotor ? :grinning:

mod38, 2019-02-16:
Phaseenablemotor still doesn't allow frequency choice as far as I can see. I'm getting better at reading source though!

lurch, 2019-02-16:
Did you try the code-snippet I pasted above? I've not tried it myself but I can't see any reason why it wouldn't work?

(Similarly, with the PhaseEnableMotor you should be able to do e.g. `motor.enable_device.frequency = 10` )

mod38, 2019-02-17:
Hi Andrew
I didn't realise that the snippet should actually work, I thought it was a suggested future format. I don't have a motor hooked up at the moment but I'll get out my Bitscope and see what is happening on the pins with and without that code. I'll report back, probably tomorrow.

mod38, 2019-02-17:
That works perfectly thank you! I put the scope across GPIO17 and GND and got 10% pulses at 10Hz. I then changed it to 50 and got pulses at 50Hz. 
I have tried to post an image of the output below.
[image]

closed 2019-02-20 by mod38

lurch, 2019-02-21:
@mod38 Glad to hear that it works okay :)

@bennuttall @waveform80 Maybe this is worth adding to the FAQ or recipes or something?

theHilikus, 2022-02-22:
Thanks for the workarounds. They work perfectly for both `Motor` and `PhaseEnableMotor`. I still believe that the right OOP approach would be to expose the parent class' attribute (`frequency`) in the child's constructor. Changing the frequency in PWM is pretty important, having to end up in this ticket to know how to do it is too cumbersome

lurch, 2022-02-22:
@theHilikus This issue is *long* closed, so in order that your reply / request doesn't get lost, you should probably open a new issue, and feel free to link to this issue.