gpiozero/gpiozero #751: Provide a method of setting per-device init params to composite devices

bennuttall, 2019-02-14:
Currently, it's not possible to provide init params to devices within compsite devices, for example:

```python
>>> bz = TonalBuzzer(20, octaves=2)
>>> bz.octaves
2
>>> bz.close()
>>> hat = JamHat()
>>> hat.buzzer.octaves
1
```

As you can see the JamHat's buzzer only has one octave, and it's not possible to change it without reconstructing the class.

@tuftii also raised this in #750.

So, rather than provide all sub-devices' init params in the composite device's init, I wonder if we could allow dictionaries of args, for example:

```python
buzzer_args = {
    'initial_value': 1,
    'mid_tone': 440.0,
    'octaves': 2,
}
hat = JamHat(buzzer_args=buzzer_args)
```

where the init param would match the name of the sub-device. Other JamHat ones would be `lights_1_args`, `lights_2_args`, `button_1_args` and `button_2_args`. I was going to exclude the buttons from that list but remembered `hold_time` is configurable - and should be for JamHat buttons.

I think most of the time most sub-devices wouldn't need to pass args, but for the ones you want ti be able to, you should. I don't think that's overcomplicated - but interested in other opinions and other options.

lurch, 2019-02-15:
In a way this almost contradicts what you said in https://github.com/RPi-Distro/python-gpiozero/pull/717#discussion_r250956252 ;-)

However I wonder if passing in a dict-of-dicts (rather than specifically-named keywords) might be a nice generic solution here? Something like:
```
buzz_args = {
    'initial_value': 1,
    'mid_tone': 440.0,
    'octaves': 2,
}
hat = JamHat(device_args={'buzzer': buzz_args})
```
That obviously works for passing in keyword-args, but what if you want to pass in positional args too? Do  we need to pass through separate buzzer_args and buzzer_kwargs parameters? (or should we keep it simple and only allow kwargs?)

Would any args passed in like this **always** override whatever args the CompositeDevice was originally trying to set? (and again, what impact does positional args have here?)

For things like the button hold_time, I guess 99% of the time you'd want the same hold_time on both of the buttons? \*cough\* buttonboard \*cough\* :wink: 

I don't have the energy to think about this in detail ATM, so just throwing out some quick thoughts.

bennuttall, 2019-02-15:
> I don't have the energy to think about this in detail ATM, so just throwing out some quick thoughts.

Same. It's not an easy one, I just wanted to start the discussion and you've already opened it up, so that's good.

lurch, 2019-02-16:
On a side-note, is it inconsistent with the rest of the GPIO Zero API that a TonalBuzzer's `.mid_tone` and `.octaves` properties can't be modified after construction? ... Or perhaps not, as e.g. `Servo` and `AngularServo` also have a bunch of properties which are set at init-time and can be read but not modified afterwards. :thinking: :man_shrugging: