gpiozero/gpiozero #756: Unable to parse new-style revision "a020d3"

AmrMohamed96, 2019-02-28:
Operating system: Ubuiquity Robotics Ubuntu Xenial
Pi model: Pi 3 Model B +
GPIO Zero version: e.g. 1.3.1

Whenever I try to run any program that uses gpiozero I am presented with an error saying "Unable to parse new-style revision "a020d3" "

Tried sudo apt install python-gpiozero but it says it is already the newest version (1.3.1~xenial.0)

AmrMohamed96, 2019-02-28:
running cat /proc/cpuinfo | tail -3

Hardware : BCM2835
Revision: a020d3

bennuttall, 2019-02-28:
The package in Ubuntu is outdated. It's up-to-date in Raspbian, and on PyPI. I'd suggest you install with pip to get the latest version:

```
sudo pip3 install gpiozero
```

@waveform80 can we get gpiozero 1.5 into Ubuntu?

Closing this as the revision code you mentioned works on more recent releases.

closed 2019-02-28 by bennuttall

lurch, 2019-02-28:
https://www.raspberrypi.org/blog/gpio-zero-v1-5/

bennuttall, 2019-02-28:
Mind you, xenial is nearly 3 years old now. I know Ubuntu are working on newer releases for the Pi, so hopefully we'll be able to keep it up to date.

lurch, 2019-02-28:
@bennuttall Looks like "Ubuiquity Robotics Ubuntu Xenial" is a 3rd-party spinoff distro.
https://downloads.ubiquityrobotics.com/pi.html
I wondered why they were still releasing new images in 2019 based off Ubuntu 16.04, and digging into it a bit further it seems they're based in turn off _another_ Ubuntu spinoff https://ubuntu-pi-flavour-maker.org/
Which seems like a dead project with no blog updates since 2017 https://ubuntu-pi-flavour-maker.org/blog/
:confused:   (which I guess in turn probably means there's no-one updating the relevant repos? Which seems to fit, given that GpioZero v1.3.1 was released in 2016 https://github.com/RPi-Distro/python-gpiozero/releases )