gpiozero/gpiozero #758: Servo docs don't make it clear that it can be driven to any position

lurch, 2019-03-20:
While looking at https://gpiozero.readthedocs.io/en/stable/api_output.html#servo it just occurred to me that if those docs are read in isolation (as a newcomer to GpioZero might do), it's not immediately obvious that the servo's `.value` can be set to _any_ floating-point number between -1 and +1. A naive beginner might assume that `.min()`, `.mid()` and `.max()` are the _only_ positions that a servo can be moved to? :man_shrugging:
Maybe I'm just taking devil's-advocate to the extreme, but I think either a tweak to the provided example-code or some clarification to the `.value` documentation would clear this up?

bennuttall, 2019-05-18:
Added an example to the [docstring](https://gpiozero.readthedocs.io/en/latest/api_output.html#servo) and more to the [recipes](https://gpiozero.readthedocs.io/en/latest/recipes.html#servo) page :+1:

closed 2019-05-18 by bennuttall

lurch, 2019-05-18:
Nice! :+1:  I just fixed your typo :)

bennuttall, 2019-05-19:
I saw - thanks

commit 23ec8842 (2019-05-18): Add more servo examples, close #758