gpiozero/gpiozero #759: Question: multiprocessing and gpiozero

judalvarezca, 2019-03-20:
Operating system: Raspbian Stretch Lite
Pi model: e.g. Pi 3 Model B+
GPIO Zero version: stable
Pin factory used: None

Good afternoon,

I'm creating a service using Gpiozero and multiple data inputs over local network. I'm trying to create multiple processes from `multiprocessing.Proccess` package within the same python script, however I can't manage it to function properly. Here is my defined class for a PIR sensor:

```python
from gpiozero import MotionSensor
import logging
import multiprocessing

class PirProcess(multiprocessing.Process):
    def __init__(self, processID, pirPin):
        logging.debug("Initializing PirThread with id: %s" % processID)
        multiprocessing.Process.__init__(self)
        self.threadID = processID
        self.pir = MotionSensor(pirPin)
        self._stop_event = multiprocessing.Event()
        logging.info("PIR System initialized.")
        
    def stop(self):
        logging.debug("Exiting PirProcess with id: %s" % self.threadID)
        self._stop_event.set()
        
    def stopped(self):
        return self._stop_event.is_set()
    
    def notifyMotion(self):
        logging.info("Motion was detected.")
        pass
    
    def notifyNoMotion(self):
        logging.debug("Motion cleared.")
        pass
    
    def run(self):
        
        self.pir.when_motion = self.notifyMotion
        self.pir.when_no_motion = self.notifyNoMotion   
            
        while not self.stopped():
            pass
        
```
The idea here is allow the final frontend user into my network to spawn and remove processes from all the security system, emulating some kind of "silence" or "activate" state for all the PIR zones.
I'm using an infinite while instead `pause()` method, already tested into the example script, allowing me to kill the process from a `SIGINT` signal. However, it never launches the PIR events `notifyMotion` and `notifyNoMotion`. Not a single time. Is `gpiozero` not compatible with the `multiprocessing` module? I'll appreciate some tips or ideas.

Thank you, and sorry if this is not the place for this question, please let me know where to find some information. Regards.

lurch, 2019-03-21:
This definitely sounds like a question for @waveform80 ..... :wink: 

But I know that GpioZero uses a lot of backgrorund threads, so it wouldn't surprise me if it's not compatible in some way with multiprocessing. :man_shrugging: 
Have you tried changing your `while` loop to do `sleep(0.01)` instead of `pass`? (perhaps the busy-loop is stealing all the CPU cycles?)

I'm not sure exactly what your code is trying to do - perhaps if you gave more context we could suggest ways of doing what you want to achieve, using the functionality built into GpioZero?

judalvarezca, 2019-03-21:
Thanks for your response!

No, I haven't tried it, but I have actually tried it with `threading` instead of `multiprocessing`. It works like a charm, no big changes in the code. That's why I'm asking for the multiprocessing compatibility.

The project is quite big, I'm building a sensitive/responsive security system, using cameras, multiple sensors  (mostly PIRs, but want to include other things like smoke or gasses, temperature, microphone, vibration, etc)  and actuators (LEDs, IR LEDs, turn ON/OFF the lights, alarm, rele drived devices, etc). I already have the cameras streaming and detecting motion using the [motion daemon](https://motion-project.github.io/), and I'm adding my own services for the sensors and actuators. I'm managing all with RPis as intermidiate servers and signal collectors, all of them connected to my main server which manages all the REST API. And then I found this ridiculously easy to use and complete tool for the RPi, and gave it a shot :wink:. 

I wanted some kind of main program to put all together: initialize all the processes and services, monitorizing each of my services and the motion service, try to respawn something if anything is going bad, report to the main server all the info, allow communication between all of the services, make logs of the valuable info for each of them, etc; instead of the actual working system (spawning like 10 consoles and running each python script in each :sweat_smile:).

lurch, 2019-03-21:
Hmmm, given GpioZero's use of background threads I'm still not sure you need to use multiprocessing for this.

This is entirely untested, but would something like the below achieve what you want?
```python
from gpiozero import MotionSensor
import logging

class PirWrapper:
    def __init__(self, processID, pirPin):
        logging.debug("Initializing PirWrapper with id: %s" % processID)
        self._ID = processID
        self.pir = MotionSensor(pirPin)
        logging.info("PIR System initialized.")
        
    def stop(self):
        logging.debug("Stopping PirWrapper with id: %s" % self._ID)
        self.pir.when_motion = None
        self.pir.when_no_motion = None
        
    def stopped(self):
        return self.pir.when_motion is not None
    
    def notifyMotion(self):
        logging.info("Motion was detected.")
    
    def notifyNoMotion(self):
        logging.debug("Motion cleared.")
    
    def run(self):
        self.pir.when_motion = self.notifyMotion
        self.pir.when_no_motion = self.notifyNoMotion   
```

closed 2021-03-12 by bennuttall