gpiozero/gpiozero #760: Add support for 5 channels line follower detector

imanousar, 2019-03-22:
Should a new class be created for line followers with 5 channels like this [one](https://www.adrirobot.it/sensori/FT-TIS-5CH/sensore_FT-TIS-5CH.htm) ?

lurch, 2019-03-22:
Yeah, I guess you could create a [CompositeDevice](https://gpiozero.readthedocs.io/en/stable/api_boards.html#compositedevice) of 5 [LineSensors](https://gpiozero.readthedocs.io/en/stable/api_input.html#linesensor-trct5000).
See also the discussion in #109

bennuttall, 2019-03-24:
```python
class CompositeLineSensor(CompositeDevice):
    def __init__(self, *pins):
        sensors = (LineSensor(pin) for pin in pins)
        super(CompositeLineSensor, self).__init__(*sensors)

>>> line_sensors = CompositeLineSensor(10, 11, 12, 13, 14)
>>> tuple(line_sensors.value)
(1, 1, 1, 1, 1) # no sensors detecting lines
>>> tuple(line_sensors.value)
(0, 1, 1, 1, 0) # leftmost and rightmost sensors detecting lines
>>> tuple(line_sensors.value)
(0, 0, 0, 0, 0) # all sensors detecting lines
```

*Note: for some reason, `LineSensor` values are the opposite to what you'd expect. 0 means line, 1 means no line.*

What would your use case be? I'd be interested to know what methods/properties you'd like to access on such a composite device, or how you'd read the values to decide what action to take.

This is very relevant to the discussion in https://github.com/RPi-Distro/python-gpiozero/issues/522#issuecomment-476001191

imanousar, 2019-03-28:
We made a line following robot for the scope of a project and we needed to read values from the [sensor](https://www.adrirobot.it/sensori/FT-TIS-5CH/sensore_FT-TIS-5CH.htm) I mentioned above. We used these values with a PID controller to fix the movement of the robot.