gpiozero/gpiozero #763: Can we expose all pin classes in gpiozero.pins?

bennuttall, 2019-04-01:
Can we move the contents of `gpiozero/pins/__init__.py to gpiozero/pins/pins.py`, and use `gpiozero/pins/__init__.py` to expose pin and factory classes like `MockFactory`, `MockPin`, `PiGPIOFactory`, etc as well as `pi_info`, so that you can import them easily in one line:

```python
>>> from gpiozero import *
>>> from gpiozero.pins import MockFactory, MockPWMPin, PiGPIOFactory
>>> Device.pin_factory = MockFactory(pin_class=MockPWMPin)
>>> led = PWMLED(2)
>>> Device.pin_factory = PiGPIOFactory(host='192.168.1.5')
>>> btn = Button(2)
```

Any reason not to?

waveform80, 2021-03-12:
The pin implementations at the moment rely on importing their underlying library and throwing ImportError in the case that it doesn't exist (because it's not installed / not supported / etc). This change would require that we come up with some other mechanism for dealing with failed imports, either catching the ImportError for each import in the `pins/__init__.py` module, or lazily importing the underlying libraries in the factory constructors. Either is a bit ugly, IMO.

Is there any particular benefit to having all these factories available straight from the pins module?

bennuttall, 2021-03-12:
No, just simpler imports. Meh, let's close it.

closed 2021-03-12 by bennuttall