gpiozero/gpiozero #766: Incorrect link in documentation

jonwitts, 2019-05-02:
On this page of of the documentation: https://gpiozero.readthedocs.io/en/stable/recipes.html#push-button-stop-motion the link to the "Push Button Stop Motion" RPi project is incorrect. It should go here: https://projects.raspberrypi.org/en/projects/push-button-stop-motion 

Thanks,
Jon

lurch, 2019-05-02:
Oooh, well spotted! :+1: 

@bennuttall Should all the other `https://www.raspberrypi.org/learning/....` links also be updated to the new `https://projects.raspberrypi.org/en/projects/....` format? :man_shrugging:

bennuttall, 2019-05-14:
Not a big deal as redirects are in place. We can update them but unless they're broken it doesn't really matter.

bennuttall, 2019-05-18:
Fixed, thanks :+1:

commit 70c53929 (2019-05-18): Update docs links to /learning projects, close #766
closed 2019-05-18 by bennuttall