gpiozero/gpiozero #767: LED.source expects iterable object

sh3ll3y, 2019-05-12:
Pi model: e.g. Pi 3 Model B
GPIO Zero version: e.g. 1.4.1
Reference: https://gpiozero.readthedocs.io/en/stable/recipes.html#internet-connection-status-indicator
Here, led.source expects an iterable object whereas a non-iterable object is assigned. There are other places also in the same page with this issue. 

```python
from gpiozero import LED, PingServer
from gpiozero.tools import negated
from signal import pause

green = LED(17)
red = LED(18)

google = PingServer('google.com')

green.source = google  # Should this be google.values?
green.source_delay = 60
red.source = negated(green)

pause()
```

lurch, 2019-05-12:
See #698 :wink: 
There was a change to the way source/values can be used, so if you upgrade to GpioZero v1.5.0 then the example code you've got should work fine.

Or if you don't want to upgrade, I think you can get the code working by changing `green.source = google` to `green.source = google.values`, and changing `red.source = negated(green)` to `red.source = negated(green.values)`

sh3ll3y, 2019-05-12:
Ah, great! Thanks  for the links :) Didn't know the change in the latest version.

closed 2019-05-12 by sh3ll3y

bennuttall, 2019-05-13:
Glad that was resolved quickly. Read about it here: https://gpiozero.readthedocs.io/en/stable/source_values.html