gpiozero/gpiozero #768: 'active_state' parameter - misbehaving

TonyHSmith, 2019-05-22:
I get an unexpected error when using the 'active_state' parameter in the InputDevice class.

Please go here: https://www.raspberrypi.org/forums/viewtopic.php?f=32&t=241064

for full details.

Many thanks.

lurch, 2019-05-22:
As @DougieLawson says, it sounds like you've got an old version of GpioZero - see https://gpiozero.readthedocs.io/en/stable/faq.html#how-can-i-tell-what-version-of-gpiozero-i-have-installed for more info about that.

And the error Dougie gets is because `.wait_for_active()` is only available on [DigitalInputDevice](https://gpiozero.readthedocs.io/en/stable/api_input.html#digitalinputdevice) or descendants, not on the base InputDevice class.

TonyHSmith, 2019-05-23:
Hi Andrew,

 

Many thanks for your prompt response – much appreciated.

 

I can confirm that I’m using V1.5.0 which I believe is the latest version. I think my problem is not down to the version, but rather the fact that I was using .wait_for_active() with the base InputDevice class, whereas (as you’ve kindly clarified) that is only available with the DigitalInputDevice class (or descendants). When I use is_active  or value everything works fine.

 

Many thanks for clearing this up for me.

 

Kind regards,

 

Tony.

 

This email has been checked for viruses by Avast antivirus software.
https://www.avast.com/antivirus

closed 2019-05-23 by bennuttall