gpiozero/gpiozero #770: DistanceSensor (HC-SR04) doesn't work with RaspPI 3 B+

francescoatmitie, 2019-05-25:
Operating system: eRaspbian Stretch
Pi model: Pi 3 Model B
GPIO Zero version: 1.5.0
Pin factory used: RPiGPIO

I've connected a HC-SR04 to a new RaspPI 3 B+ and used the simplest example given for DistanceSensor.
The script runs but sometimes nothing is output, sometimes it starts and set the distance to maximum, or prints a warning.

So I decided to use pure GPIO code example I found in the internet and works with no problems.
Here what I use (it runs in a thread):

```python

#!/usr/bin/python3

import RPi.GPIO as GPIO
import os, sys, time, datetime, _thread

# exit flag to stop threads
exitFlag = False

hc_trigger_gpio = 5
hc_echo_gpio    = 6

#
# Distance sensor management
#

def read_distance_sensor(threadName):

    global exitFlag

    # alarm status
    alarm_status = False

    # How long to wait (in seconds) between measurements
    frequency_seconds = 1

    # GPIO Mode (BOARD / BCM)
    GPIO.setmode(GPIO.BCM)

    # config HC-SR04 ultrasonic distance sensor GPIO
    # set GPIO direction (IN / OUT)
    GPIO.setup(hc_trigger_gpio, GPIO.OUT)
    GPIO.setup(hc_echo_gpio, GPIO.IN)

    def measure_distance():

        # set Trigger to HIGH
        GPIO.output(hc_trigger_gpio, True)

        # set Trigger after 0.01ms to LOW
        time.sleep(0.00001)
        GPIO.output(hc_trigger_gpio, False)

        StartTime = time.time()
        StopTime = time.time()

        # save StartTime
        while GPIO.input(hc_echo_gpio) == 0:

            StartTime = time.time()

        # save time of arrival
        while GPIO.input(hc_echo_gpio) == 1:

            StopTime = time.time()

        # time difference between start and arrival
        TimeElapsed = StopTime - StartTime

        # multiply with the sonic speed (34326 cm/s)
        # and divide by 2, because there and back
        return ((TimeElapsed * 34326) / 2)

    while exitFlag is False:

        distance = measure_distance()

        print('Distance: {0:0.2f} cm'.format(distance))

        time.sleep(frequency_seconds)

    print("Exit {0} thread".format(threadName))

def main():

    global exitFlag

    # Printout startup string
    print('Press Ctrl-C to quit.')

    # Create threads to monitor the sensors
    try:

        _thread.start_new_thread(read_distance_sensor, ("Thread-HC", ))

    except:

        print("Error: unable to start the threads")

    try:

        # Endless main loop
        while True:

            pass

    except KeyboardInterrupt:

        print('Keyboard Interrupt')

        # Notify threads it's time to exit
        exitFlag = True

        # give some time for the threads to exit
        time.sleep(5)

        print('Exit main')

if __name__ == '__main__':

    main()
```

It looks to me like DistanceSensor in 1.5.0 is still not working as expected.

bennuttall, 2019-05-26:
Have you tried using PiGPIO Factory as recommended in the docs? https://gpiozero.readthedocs.io/en/stable/api_input.html#distancesensor-hc-sr04

francescoatmitie, 2019-06-04:
I didn't try and I didn't have the time to, so I stick to what I implemented above as that will do.
I don't need great precision, just make sure that if something is closer than 20cm an alert is triggered.

bennuttall, 2019-06-11:
Can you post your gpiozero code that doesn't work?

closed 2019-09-20 by bennuttall