gpiozero/gpiozero #772: TimeOfDay: Multiple timeframes possible?

tibaultbordeaux, 2019-06-06:
Operating system: Raspbian Stretch
Pi model: Pi Zero W
GPIO Zero version: latest
Pin factory used: e.g. RPiGPIO

Hi folks, thanks for this wonderful piiece oft software.

I'm using TimeOfDay() for switching a relay.
Now I'm wondering how I would program a schedule with two or more timeframes on a single day.

Like:

```python
import gpiozero
from datetime import time
from gpiozero import TimeOfDay

Relay = gpiozero.OutputDevice(23)

delta1 = TimeOfDay(time(8,15), time(8,30), False)
delta2 = TimeOfDay(time(21,45), time(22,0), False)

Relay.source = delta1
Relay.source = delta2
```

bennuttall, 2019-06-06:
How about:

```python
from datetime import time
from gpiozero import TimeOfDay, OutputDevice
from gpiozero.tools import any_values

relay = OutputDevice(23)

delta1 = TimeOfDay(time(8, 15), time(8, 30), utc=False)
delta2 = TimeOfDay(time(21, 45), time(22, 0), utc=False)

relay.source = any_values(delta1, delta2)
```

`any_values` is essentially "or". Likewise, `all_values` is essentially "and".

See https://gpiozero.readthedocs.io/en/stable/source_values.html and https://gpiozero.readthedocs.io/en/stable/api_tools.html#module-gpiozero.tools

tibaultbordeaux, 2019-06-07:
Thanks, thats great!

Would it even be possible to combine these two approaches like this:

```Python
from datetime import time
from gpiozero import TimeOfDay, OutputDevice, DistanceSensor
from gpiozero.tools import any_values, all_values

relay = OutputDevice(23)
sensor = DistanceSensor(14, 15)

def checkDistance:
     if sensor.distance < 10
          return True

delta1 = TimeOfDay(time(8, 15), time(8, 30), utc=False)
delta2 = TimeOfDay(time(21, 45), time(22, 0), utc=False)

relay.source = all_values(any_values(delta1, delta2), checkDistance)

```

bennuttall, 2019-06-07:
Yes, you can combine them but that example won't work as `checkDistance` is a function. You want to use an iterator, like a generator function:

```python
def check_distance():
    while True:
         if sensor.distance < 10
              yield True
        else:
            yield False

distance = check_distance()

relay.source = all_values(any_values(delta1, delta2), distance)
```

However, you don't need to do that, as you can use the `DistanceSensor` object itself. You just need to booleanize its value (which is 0->1 on a scale from 0m to `max_distance`):

```python
from datetime import time
from gpiozero import TimeOfDay, OutputDevice, DistanceSensor
from gpiozero.tools import any_values, all_values, booleanized

relay = OutputDevice(23)
distance = DistanceSensor(14, 15, max_distance=10)

delta1 = TimeOfDay(time(8, 15), time(8, 30), utc=False)
delta2 = TimeOfDay(time(21, 45), time(22, 0), utc=False)

relay.source = all_values(any_values(delta1, delta2), booleanized(distance, 1, 2))
```

Here I set the distance sensor's `max_distance` to 10. That means a distance of 10m (or higher) returns a value of 1.0, a distance of 5m returns a value of 0.5. Since you want to trigger the relay if it's over 10m, you only care if the sensor reports a value of 1.0, so booleanizing it (between 1 and (somewhat arbitrary) 2) turns anything other than 1.0 into False.

Does that make sense?

lurch, 2019-06-07:
Another example of where #589 would come in handy :wink:

tibaultbordeaux, 2019-06-09:
Dear @bennuttall ,

that makes totally sense. I'm amazed how gpiozero combines simplicity und functionality. Thanks a lot for your detailed help.

Despite of becoming slightly off-topic: The distance sensor was used as an example. In fact I'm trying to build an automated watering machine for my plants. The relay is switching on a pump, once in the morning and once in the evening. But only if there is not enough soil moisture. Therefor I want to use a Xiaomi Mi Flora Sensor by importing [https://github.com/open-homeautomation/miflora/](https://github.com/open-homeautomation/miflora/)

Now I know how to do that with gpiozero:

```Python
from datetime import time
from gpiozero import TimeOfDay, OutputDevice, DistanceSensor
from gpiozero.tools import any_values, all_values, booleanized
from miflora.miflora_poller import MiFloraPoller
from btlewrap.bluepy import BluepyBackend

poller = MiFloraPoller('xx:xx:xx:xx:xx:xx', BluepyBackend)
moisture = poller.parameter_value('moisture') # moisture in %

relay = OutputDevice(23)

delta1 = TimeOfDay(time(8, 15), time(8, 30), utc=False)
delta2 = TimeOfDay(time(21, 45), time(22, 0), utc=False)

relay.source = all_values(any_values(delta1, delta2), booleanized(moisture, 0, 30))
```

This way my plants get watered only if the soil moisture is below 30%.

Thanks again!

bennuttall, 2019-06-09:
That's great to hear! I often wonder whether people progress onto these more advanced features - glad you've discovered this style of programming :)

Two ways, either write a generator function which yields boolean values:

```python
def moisture_values():
    while True:
        yield poller.parameter_value('moisture') < 30

moisture = moisture_values()

relay.source = all_values(any_values(delta1, delta2), moisture)
```

or create a class that extends a GPIO Zero device:

```python
from gpiozero import ValuesMixin

class MoistureSensor(ValuesMixin):
    @property
    def value(self):
        return poller.parameter_value('moisture') / 100

moisture = MoistureSensor()
relay.source = all_values(any_values(delta1, delta2), booleanized(moisture, 0, 0.3))
```

closed 2019-06-11 by bennuttall

tibaultbordeaux, 2019-06-23:
Thanks @bennuttall , your last addition was a great help, things are working now.
(I choosed the first approach: your generator function)
May I add another question?

My Pi is solar-/battery-powered and therefor I would like to save energy whereever possible.
Do I understand it correctly, that your code above is constantly checking the moisture values?
Since it's a bluetooth connection I would like to avoid this.

The moisture data should only be pulled once from the sensor when the timeframes (delta1, delta2) start.
Is there any thing in  GPIO Zero I could use for that?

Thanks  in advance!

bennuttall, 2019-06-23:
Using `source_delay` will allow you to pause between checking. The default is 0.01 seconds.

```python
relay.source = 60  # check once per minute
relay.source = all_values(any_values(delta1, delta2), booleanized(moisture, 0, 0.3))
```

tibaultbordeaux, 2019-06-23:
Yes, I've seen this in the docs. Still the sensor values are checked more often than necessary, but I think it's okay. Thank you, Ben!

bennuttall, 2019-06-23:
Yeah, that's the case with the pull-based approach we use. If you could convert it to be push-based, you'd be able to get around this.

Here's an example where I wrap the `Button` class to only update `value` when it changes:

```python
from gpiozero import Button, LED, ValuesMixin

class MyButton(ValuesMixin):
    def __init__(self, pin):
        self.btn = Button(pin)
        self.value = 0
        self.btn.when_pressed = self.pressed
        self.btn.when_released = self.released

    def pressed(self):
        self.value = 1

    def released(self):
        self.value = 0

led = LED(2)
btn = MyButton(3)

led.source = btn
```

So if you had a way of updating a variable when your sensor changes, you could write something (a class or function) like this. Does that make sense?

tibaultbordeaux, 2019-06-24:
Thanks again. To be honest: Currently the creating of such a class is beyond my coding skills. But I'll come back after this bug is fixed ...