gpiozero/gpiozero #774: Can PCF8591 be supported?

jikoy, 2019-06-20:
Can PCF8591 be supported?

bennuttall, 2019-06-20:
More information please. What is it? What's it for? What protocol does it use? Have you got exiting code to interface with it?

jikoy, 2019-06-20:
I seem to have found a solution. Thank you.

jikoy, 2019-06-21:
import smbus
import time

# for RPI version 1, use "bus = smbus.SMBus(0)"
bus = smbus.SMBus(1)

#check your PCF8591 address by type in 'sudo i2cdetect -y -1' in terminal.
def setup(Addr):
	global address
	address = Addr

def read(chn): #channel
	if chn == 0:
		bus.write_byte(address,0x40)
	if chn == 1:
		bus.write_byte(address,0x41)
	if chn == 2:
		bus.write_byte(address,0x42)
	if chn == 3:
		bus.write_byte(address,0x43)
	bus.read_byte(address) # dummy read to start conversion
	return bus.read_byte(address)

def write(val):
	temp = val # move string value to temp
	temp = int(temp) # change string to integer
	# print temp to see on terminal else comment out
	bus.write_byte_data(address, 0x40, temp)

if __name__ == "__main__":
	setup(0x48)
	while True:
		print 'AIN0 = ', read(0)
		print 'AIN1 = ', read(1)
		tmp = read(0)
		tmp = tmp*(255-125)/255+125 # LED won't light up below 125, so convert '0-255' to '125-255'
		write(tmp)
#		time.sleep(0.3)

jikoy, 2019-06-21:
#!/usr/bin/env python
import PCF8591 as ADC

def setup():
	ADC.setup(0x48)

def loop():
	while True:
		print ADC.read(0)
		ADC.write(ADC.read(3))

def destroy():
	ADC.write(0)

if __name__ == "__main__":
	try:
		setup()
		loop()
	except KeyboardInterrupt:
		destroy()

jikoy, 2019-06-21:
How can I rewrite it with GPIOZERO?

lurch, 2019-06-21:
GpioZero doesn't support I2C yet.

jikoy, 2019-06-21:
Is there a plan?

closed 2019-09-20 by bennuttall