gpiozero/gpiozero #775: Time based code execution possible?

tibaultbordeaux, 2019-06-23:
Operating system: Raspbian Stretch
Pi model: Pi Zero W
GPIO Zero version: latest

I like the possibilities of TimeOfDay. This is for doing things BETWEEN two dates.
Is there a similar solution to achieve a code execution AT a specific point in time?
And a similar solution to do something on a regular basis in  a specific time INTERVALL?

Something like:
```python
TimeInterval(time(8, 45), 5, 12)
```
for starting at 8:45 every 5 minutes for 12 times.

```python
TimeInterval(time(8, 45), 5, 0)
```
for doing it endless every 5 minutes

```python
TimeInterval(time(8, 45), 0, 1)
```
for doing it once

Sure, I could achieve this with a proper cron entry but I was wondering if I could do it programmatically with GPIO Zero.

Thanks!

bennuttall, 2019-06-23:
There could be. There's an outstanding issue with `TimeOfDay` which if fixed, would allow this.

With input devices, you have four options for reading it: polling (`if btn.is_pressed`), blocking (`btn.wait_for_press()`), callback (`btn.when_pressed = x`) and delcarative (`x.source = btn`).

With internal devices, `when_activated` and `when_deactivated` properties exist, but they're not implemented. The same goes for `wait_for_activated` - it just won't work.

So in theory, you can use:

```
from gpiozero import TimeOfDay
from datetime import time

tod = TimeOfDay(time(8, 45))

tod.when_activated = lambda: print("activated")
```

See issue #393

tibaultbordeaux, 2019-06-24:
Thanks for pointing out. I'll wait for the fix. :-)

closed 2019-09-20 by bennuttall