gpiozero/gpiozero #782: How do I increase the baud rate of the clock pin for SPI?

Mockapapella, 2019-07-26:
Operating system: e.g. Raspbian Stretch
Python version: 3.5.3
Pi model: Raspberry Pi Zero W Rev 1.1 and 
GPIO Zero version: 1.5.0
Pin factory used: <gpiozero.pins.rpigpio.RPiGPIOFactory object at 0xb560ae70>

When run, this program monitors 3 channels on an MCP3204, 2 for volts and another for amps (by using a function to convert the measured volts to amps). It uses those values to calculate a cost projection and watts, then adds those to a list to average out. It keeps a running list of the last 100 values to average out. It also keeps track of how many times it has sampled in the last second, and generates a call graph image to help visualize what classes and being called when `pin_0` is being initialized. As of now it is measuring at ~270-275 samples per second on a Pi Zero and ~1335-1340 samples per second for a Pi 3 B+ (All other factors remaining the same).

While looking for a way to access the baud rate, I can across [this line](https://github.com/RPi-Distro/python-gpiozero/blob/master/gpiozero/pins/pigpio.py#L368). According to the [call graph](https://i.imgur.com/CLHzj8G.jpg), I'm pretty sure when initializing the `pin_0` object, that baud rate line is reference. Another supporting factor is that when measuring the clock pin with an oscilloscope, it is sitting at 500kHz. Despite this, no matter what I try, I can't figure out how to change it.

```
import time
import gpiozero
import numpy
from pycallgraph2 import Config as GraphConfig
from pycallgraph2 import PyCallGraph
from pycallgraph2.globbing_filter import GlobbingFilter
from pycallgraph2.output import GraphvizOutput

graph_output = GraphvizOutput()
# graph_output.should_update = True
graph_output.output_file = "basic.png"

graph_configuration = GraphConfig()
graph_configuration.trace_filter = GlobbingFilter(exclude=["kivy.*", "pycallgraph2.*"])
graph_configuration.groups = False
graph_configuration.include_stdlib = True
# graph_configuration.should_update = True

def get_volts(raw_channel_0_voltage, raw_channel_1_voltage, previous_channel_read):

    v_in = (raw_channel_1_voltage / 1000) * ((6.04 + 270) / 6.04)
    if previous_channel_read == 0:
        if v_in < 48:
            return (raw_channel_0_voltage / 1000) * ((24 + 270) / 24), 0
        else:
            return (raw_channel_1_voltage / 1000) * ((6.04 + 270) / 6.04), 1
    elif previous_channel_read == 1:
        if v_in > 46:
            return (raw_channel_1_voltage / 1000) * ((6.04 + 270) / 6.04), 1
        else:
            return (raw_channel_0_voltage / 1000) * ((24 + 270) / 24), 0

def get_amps(raw_channel_current):
    i_in = (raw_channel_current / 1000) * 4
    if i_in > 0:
        # return current_correction_function(i_in)
        return i_in
    elif i_in <= 0:
        return 0.0

def kilowatt_total(watts):
    bump_factor = 1.7
    kilowatt_hour = 0.12
    hours_in_a_month = 300
    months = 12
    number_of_panels = 4
    watt_to_kilowatt = 1000
    flourescent_to_fusion_multiplier = 1.83

    return (
        (((watts * bump_factor * kilowatt_hour / watt_to_kilowatt) * hours_in_a_month) * months)
        * number_of_panels
    ) * flourescent_to_fusion_multiplier

def current_correction_function(amps):
    corrected_value = (amps / 1.0536) + 0.0129
    return corrected_value

with PyCallGraph(output=graph_output, config=graph_configuration):
    pin_0 = gpiozero.MCP3204(
        channel=0,
        differential=False,
        max_voltage=4.096,
        clock_pin=11,
        mosi_pin=10,
        miso_pin=9,
        select_pin=8,
    )
# print(pin_0.channel.get_frequency())
pin_1 = gpiozero.MCP3204(
    channel=1,
    differential=False,
    max_voltage=4.096,
    clock_pin=11,
    mosi_pin=10,
    miso_pin=9,
    select_pin=8,
)
pin_2 = gpiozero.MCP3204(
    channel=2,
    differential=False,
    max_voltage=4.096,
    clock_pin=11,
    mosi_pin=10,
    miso_pin=9,
    select_pin=8,
)

electricity_list = numpy.array([[0, 0, 0]])
money_list = numpy.array([])
max_volts = 0
max_amps = 0
max_watts = 0
max_dollars = 0
start_time = time.time()
previous_channel_read = 0
i = 0
try:
    print(True)
    with open("values.txt", "r") as file:
        file = list(file)
        file = file[0].split(",")
        print(file)
        max_volts = float(file[0])
        print(max_volts)
        max_amps = float(file[1])
        print(max_amps)
        max_watts = float(file[2])
        print(max_watts)
        money_list = numpy.array([float(file[3])])
        print(money_list)
except Exception as e:
    print(e)

while True:
    volts, previous_channel_read = get_volts(
        pin_0.raw_value, pin_1.raw_value, previous_channel_read=previous_channel_read
    )
    amps = get_amps(pin_2.raw_value)
    watts = volts * amps

    electricity_list = numpy.append(electricity_list, [[volts, amps, watts]], axis=0)
    electricity_list = electricity_list[-100:]
    electricity_list_average = numpy.mean(electricity_list, axis=0)

    v_average = electricity_list_average[0]
    a_average = electricity_list_average[1]
    w_average = electricity_list_average[2]
    dollars = kilowatt_total(w_average)

    if volts > max_volts:
        max_volts = volts
    if amps > max_amps:
        max_amps = amps
    if watts > max_watts:
        max_watts = watts
    if dollars > max_dollars:
        max_dollars = dollars

    if time.time() > start_time + 1:
        # print("{:.2f}V ({:.2f}V), {:.2f}A ({:.2f}A), {:.2f}W ({:.2f}W)".format(volts, max_volts, amps, max_amps, watts, max_watts))
        print("{} samples in the last second".format(i))
        start_time = time.time()
        i = 0
    i += 1

    # with open("values.txt", "w+") as f:
    #     f.write("{},{},{},{}".format(max_volts, max_amps, max_watts, max_dollars))
```

joan2937, 2019-07-27:
The linked line with 500000 seems to be associated with the pigpio backend.

Pin factory used: <gpiozero.pins.rpigpio.RPiGPIOFactory object at 0xb560ae70> seems to suggest you are actually using the RPi.GPIO backend.

Mockapapella, 2019-07-29:
So does that mean that since the baud rate is never set within that pin factory, I can set it to a faster speed than 500000 before running the program and my program will use that faster speed instead? Or does it simply mean I can't boost the speed of the clock pin?

joan2937, 2019-07-29:
No.  I am just saying the line you linked seems to be associated with the pigpio backend but you seem to be using the RPi.GPIO backend.

Your question is how do I set the SPI baud rate in gpiozero.  I do not know the answer to that question.

Mockapapella, 2019-07-29:
Ah OK my mistake. Thanks for pointing that out. I will probably end up switching over to another library if I can't boost the speed up to at least a couple thousand times per second.

bennuttall, 2019-07-29:
You can use the pigpio library directly (`import pigpio`) or see if you can figure out how to do what you want by poking underneath at the pin factory object. See https://gpiozero.readthedocs.io/en/stable/api_pins.html#changing-the-pin-factory https://gpiozero.readthedocs.io/en/stable/api_pins.html#module-gpiozero.pins.pigpio and https://github.com/gpiozero/gpiozero/blob/master/gpiozero/pins/pigpio.py#L368

joan2937, 2019-07-29:
**Unless** @Mockapapella is using the remote features the best performance is likely to come from using the Python spidev module.  The spidev module does allow for setting the baud rate.  For all I know gpiozero may use spidev under the hood anyhow.  I'd use gpiozero for the code where possible and spidev to handle SPI.

bennuttall, 2019-07-29:
I believe we do, as long as it's installed and SPI is enabled. I don't think we only use it for RPi.GPIO factory. Ping @waveform80

Mockapapella, 2019-07-30:
I imagine the syntax for something like this would be similar to:
```
import spidev
import gpiozero

pin_0 = gpiozero.MCP3204(
    channel=1,
    differential=False,
    max_voltage=4.096,
    clock_pin=11,
    mosi_pin=10,
    miso_pin=9,
    select_pin=8,
)
```
And then after the object and pins are initialized, access the specific pin and frequency within the pin factory like:
```
spidev.clock_pin.baud_rate = 1000000
pin_0.pin_factory.clock_pin.frequency = spidev.clock_pin.baud_rate
```
...or something like that. I'm not too familiar with the syntactic connection between a pin/pin factory and the initialization of `pin_0 = gpiozero.MCP3204(...)`.

lurch, 2019-07-30:
See also #650

waveform80, 2021-03-12:
Fixed in b30983a1aeec9d83af555394c663c965761ac4b6

closed 2021-03-12 by waveform80

slim-yeti, 2024-01-20:
This worked  for me to reduce speed.
rom gpiozero import MCP3008

#create an object called pot that refers to MCP3008 channel 0
rate=10000
pot0 = MCP3008(0)
print(pot0._spi._get_rate())
pot0._spi._set_rate(rate)
print(pot0._spi._get_rate())