gpiozero/gpiozero #783: RuntimeError when calling RPi.GPIO.cleanup()

attwoodn, 2019-07-27:
Operating system: Raspbian GNU/Linux 10 (buster)
Python version: 3.7.3
Pi model: Pi 3 Model B+
GPIO Zero version: 1.5.1
Pin factory used: RPiGPIOFactory

A RuntimeError occurs every time I call RPi.GPIO.cleanup() in programs that use gpiozero objects. I've read the docs and I haven't been able to spot a place where it explicitly states _not_ to use RPi.GPIO.cleanup(), but it does point the reader to gpiozero alternatives. Of course we can avoid using RPi.GPIO.cleanup() by using the close() method on gpiozero objects instead, but I feel that using RPi.GPIO.cleanup() shouldn't result in cryptic RuntimeErrors. The problem is that I've been using other python libraries in conjunction with gpiozero, and I don't have control over how these other libraries perform their cleanup routines. As a result, I've seen this RuntimeException a few times across different programs I've written.

Here is a simple example program that causes the error:
```python
from gpiozero import Button
from RPi import GPIO

btn = Button(14)
GPIO.cleanup()
```

When the above program is run with python 3.7.3, this is the resulting stacktrace:
```
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 302, in _set_when_changed
    self._disable_event_detect()
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 253, in _disable_event_detect
    GPIO.remove_event_detect(self.number)
RuntimeError: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)
Exception ignored in: <function GPIOBase.__del__ at 0x764cc588>
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 151, in __del__
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 394, in close
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 540, in close
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 136, in close
  File "/usr/lib/python3/dist-packages/gpiozero/pins/__init__.py", line 450, in <lambda>
  File "/usr/lib/python3/dist-packages/gpiozero/pins/pi.py", line 302, in _set_when_changed
  File "/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py", line 253, in _disable_event_detect
RuntimeError: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)
```

When the above program is run with python 2.7.16, this is the resulting stacktrace:
```
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/dist-packages/gpiozero/devices.py", line 595, in _shutdown
    _devices_shutdown()
  File "/usr/lib/python2.7/dist-packages/gpiozero/devices.py", line 588, in _devices_shutdown
    dev.close()
  File "/usr/lib/python2.7/dist-packages/gpiozero/mixins.py", line 394, in close
    super(HoldMixin, self).close()
  File "/usr/lib/python2.7/dist-packages/gpiozero/devices.py", line 540, in close
    self._pin.close()
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/rpigpio.py", line 136, in close
    self.when_changed = None
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/__init__.py", line 450, in <lambda>
    lambda self, value: self._set_when_changed(value),
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/pi.py", line 302, in _set_when_changed
    self._disable_event_detect()
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/rpigpio.py", line 253, in _disable_event_detect
    GPIO.remove_event_detect(self.number)
RuntimeError: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/lib/python2.7/dist-packages/gpiozero/devices.py", line 595, in _shutdown
    _devices_shutdown()
  File "/usr/lib/python2.7/dist-packages/gpiozero/devices.py", line 588, in _devices_shutdown
    dev.close()
  File "/usr/lib/python2.7/dist-packages/gpiozero/mixins.py", line 394, in close
    super(HoldMixin, self).close()
  File "/usr/lib/python2.7/dist-packages/gpiozero/devices.py", line 540, in close
    self._pin.close()
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/rpigpio.py", line 136, in close
    self.when_changed = None
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/__init__.py", line 450, in <lambda>
    lambda self, value: self._set_when_changed(value),
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/pi.py", line 302, in _set_when_changed
    self._disable_event_detect()
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/rpigpio.py", line 253, in _disable_event_detect
    GPIO.remove_event_detect(self.number)
RuntimeError: Please set pin numbering mode using GPIO.setmode(GPIO.BOARD) or GPIO.setmode(GPIO.BCM)
```

Thank you for your hard work on this library. 
Best wishes,

Noah

bennuttall, 2019-07-28:
Don't call `GPIO.cleanup()`. You don't need to and, as you've noticed, it breaks things.

See the note in the docs: https://gpiozero.readthedocs.io/en/stable/migrating_from_rpigpio.html#cleanup

If you want to clean up a GPIO channel you've used, do it the gpiozero way, not the RPi.GPIO way:

```python
>>> led = LED(2)
>>> led.blink()
>>> led.close()
```

This will call `GPIO.cleanup(2)` underneath, when you're using the RPiGPIO pin factory, and the equivalent for other pin factories.

bennuttall, 2019-07-28:
Also note that cleanup takes place automatically at the end of a script, so only call `close()` if you need to change what you're using a pin for.

closed 2019-07-28 by bennuttall

lurch, 2019-07-30:
> The problem is that I've been using other python libraries in conjunction with gpiozero, and I don't have control over how these other libraries perform their cleanup routines.

Do you have any specific examples of other libraries that don't "play nicely" with gpiozero?

attwoodn, 2019-07-31:
I first noticed this issue when using a library called pad4pi - https://github.com/brettmclean/pad4pi . The cleanup method of the Keypad class simply calls RPi.GPIO.cleanup()

lurch, 2019-08-01:
It doesn't help your immediate problem, but pad4pi wouldn't be needed if #358 got implemented.