gpiozero/gpiozero #785: RGB fade

TGWC, 2019-07-31:
It's possible do fade a RGB trought all colours? Also, how to make a RGB blink like xmas ligths? Cannot make it work.

lurch, 2019-07-31:
> It's possible do fade a RGB trought all colours?

I actually started working on this a couple of years ago, but never got round to finishing it :slightly_frowning_face: Other things took over, and I completely forgot about it.
See #228 and #639

If I get time next week (which may be unlikely) I'll see if I can dig it out.

TGWC, 2019-07-31:
Thanks foryour reply :D 

Looking forward to see it working one day..would love to have it for my wedding in the end of August as I am doing a photobooth with PI and gpiozero and would like to have leds around it fading all colours, lol

bennuttall, 2019-07-31:
You can do that with colorzero, e.g something like this:

```python
from gpiozero import RGBLED
from colorzero import Color, Hue
from time import sleep

led = RGBLED(2, 3, 4)

led.color = Color('red')  # starting color
for i in range(360):
    led.color += Hue(deg=i)
    sleep(0.1)
```

TGWC, 2019-07-31:
> You can do that with colorzero, e.g something like this:
> 
> ```python
> from gpiozero import RGBLED
> from colorzero import Color, Hue
> from time import sleep
> 
> led = RGBLED(2, 3, 4)
> 
> led.color = Color('red')  # starting color
> for i in range(360):
>     led.color += Hue(deg=i)
>     sleep(0.1)
> ```

it's possible to make this inside a thread so it does not block the main script? Just wondering about it

TGWC, 2019-08-01:
Also this does not work led.pulse(fade_in_time=5, fade_out_time=5), it stays all black...

bennuttall, 2019-08-05:
> it's possible to make this inside a thread so it does not block the main script? Just wondering about it

Yes

closed 2019-08-05 by bennuttall