gpiozero/gpiozero #786: RGBLED

TGWC, 2019-08-01:
RGBLED methods blink and pulse do not work.

RGBled.pulse(fade_in_time=5, fade_out_time=5)

bennuttall, 2019-08-05:
Are you doing this on the object or on the class? Should be:

```python
led = RGBLED(2, 3, 4)

led.puise()
```

Do colours tend to work? Red? Blue? Green? Anything else? If a colour isn't working you need to check your wiring.

What happens when you try `led.pulse()` without any args?

How about `led.pulse(on_color=(1, 0, 0), off_color=(0, 0, 1)` for example? (changing from red to green)

TGWC, 2019-08-05:
They do not work when using Flask gevent, only if I use them with just flask...

bennuttall, 2019-08-05:
That's not going to work. You should have mentioned that in your issue. As I suggested, please try the forums for this kind of troubleshooting. It's not an issue with the library, it's a paradigm mix-up you need to learn about.

closed 2019-08-05 by bennuttall