gpiozero/gpiozero #789: MotionSensor.value does not update

vinceemail, 2019-08-05:
When running the following simple code, I would expect the [while True] to return an endless stream of 1 and 0, relating to the pir motion sensor state. However, once triggered, I only get 1s, even once motion has ended. If I do the opposite (ie, put the loop into a when_no_motion), I get a string of 0s... It seems that the pir.value is not updating. 

Any clues? 

Thanks in advance! 

```python
from signal import pause
from gpiozero import MotionSensor
pir = MotionSensor(4, queue_len=1)

def do_motion_detected():
  while True:
    print(pir.value)

pir.when_motion = do_motion_detected

pause()
```

---

Operating system: e.g. Raspbian Buster
Python version: e.g. 2.7.13
Pi model: e.g. Pi Zero W
GPIO Zero version: e.g. 1.5.1
Pin factory used: e.g. RPiGPIO

vinceemail, 2019-08-05:
It might also be worth noting that, when I try this with GPIOZero Button instead of MotionSensor, it works fine, giving me a stream of 1s and 0s, correlating to the Button value... 

```python
from signal import pause
from gpiozero import Button
clicker = Button(4)

def do_press_detected():
  while True:
    print(clicker.value)

clicker.when_pressed = do_press_detected

pause()
```

bennuttall, 2019-08-05:
Don't use a while loop inside your callback function.

Either do:

```python
while True:
    print(pir.value)
```

or:

```python
def changed(pir):
    print(pir.value)

pir.when_motion = changed
pir.when_no_motion = changed
```

vinceemail, 2019-08-05:
Thanks, that's the workaround I've found. Any idea why it works with Button, but not MotionSensor? Seems inconsistent, so I must lack knowledge.

Thanks again..

vinceemail, 2019-08-05:
Thanks, that's the workaround I've found. It's not ideal, because I'm actually running multiple sensors, and this will affect the broader code (the above examples are hugely simplified, of course). 

Still... Any idea why it works with Button, but not MotionSensor? 

Seems inconsistent, so I must lack knowledge.

Thanks again..

bennuttall, 2019-08-05:
Probably because `btn.value` is easy to calculate (just reads the GPIO pin) but motion sensor requires averaging the last few values in a queue, it can't keep up. It's just wrong to try to do that though. There's nothing to stop a callback once you've started it - so every time there's motion a new thread starts that never ends.

closed 2019-08-05 by bennuttall