gpiozero/gpiozero #790: RGBLED - Common Anode vs. Common Cathode

LongLane, 2019-08-17:
Operating system: Raspbian Buster
Python version: 3.7
Pi model: RPi Zero W
GPIO Zero version: latest
Pin factory used: e.g. RPiGPIO
See http://rpf.io/gpzissue for information on how to find out these details

Please give us as much information about your issue as possible. Write your
code inside code blocks like so:

Justs noticed while using a Common Anode RGB Led that gpiozero assumes a Common Cathode. I would be helpful if there was an additional option to specify Common Anode. E.g. Adafruit prefers selling RGB Leds based on CA.

In terms of code, it has be reversed, i.e. 0 is on, 1 is off.

Thanks

bennuttall, 2019-08-17:
```python
led = RGBLED(2, 3, 4, active_high=False)
```

https://gpiozero.readthedocs.io/en/stable/api_output.html#rgbled

LongLane, 2019-08-17:
My bad! Many thanks.

closed 2019-08-17 by LongLane