gpiozero/gpiozero #791: Motor Class does not pass pin_factory to pins.

yisraeldov, 2019-08-18:
Operating system: e.g. Raspbian Buster
Python version: e.g. 3.6
Pi model: e.g. Pi 4 Model B
GPIO Zero version: e.g. 1.5.1
Pin factory used: e.g. MockFactory
See http://rpf.io/gpzissue for information on how to find out these details

Please give us as much information about your issue as possible. Write your
code inside code blocks like so:

Here is the init for Motor
```python
def __init__(self, forward=None, backward=None, enable=None, pwm=True,
                 pin_factory=None):
        if not all(p is not None for p in [forward, backward]):
            raise GPIOPinMissing(
                'forward and backward pins must be provided'
            )
        PinClass = PWMOutputDevice if pwm else DigitalOutputDevice
        devices = OrderedDict((
            ('forward_device', PinClass(forward)),
            ('backward_device', PinClass(backward)),
        ))
        if enable is not None:
            devices['enable_device'] = DigitalOutputDevice(enable,
                                                           initial_value=True)
        super(Motor, self).__init__(_order=devices.keys(), **devices)
```
When the devices are enabled the `pin_class` is never propagated to them.

lurch, 2019-08-18:
I don't have time to test right now, but I _think_ the fix would be to change it to:
```python
        devices = OrderedDict((
            ('forward_device', PinClass(forward, pin_factory=pin_factory)),
            ('backward_device', PinClass(backward, pin_factory=pin_factory)),
        ))
        if enable is not None:
            devices['enable_device'] = DigitalOutputDevice(enable,
                                                           initial_value=True,
                                                           pin_factory=pin_factory)
        super(Motor, self).__init__(_order=devices.keys(), pin_factory=pin_factory, **devices)
```
Would you mind testing it? :slightly_smiling_face: 

ping also @waveform80 since he knows the pin_factory stuff much better than me... (I'm not sure why `CompositeDevice` also has a `pin_factory` argument?)

yisraeldov, 2019-08-18:
Yep fixed it like this before I saw your message...  Doing a PR in a few min...

@waveform80 Can you take a look please

PR #792 (2019-09-17): fixing motors to work with pin_factory
This should fix #791 

It fixes my issue at least
closed 2019-09-17 by bennuttall