gpiozero/gpiozero #793: Multiple scripts using a button on the same pin stops working when one quits

tonyyng, 2019-08-22:
Operating system: Raspbian Buster
Python version: 3.7.3
Pi model: Pi 3 Model B Rev 1.2
GPIO Zero version: 1.5.1
Pin factory used: Empty string
from gpiozero import Device
Device
<class 'gpiozero.devices.Device'>
Device.pin_factory
(blank line)

I'm trying to run multiple instances (tried two and three) of a script that uses gpiozero to read a button state.  The code is based on this example: https://www.raspberrypi.org/forums/viewtopic.php?t=188755#p1188826

```#!/usr/bin/env python3

import gpiozero
import time
import os

class cmdbutton(object):

    def __init__(self, pin, short_duration, long_duration, short_action=None, long_action=None):
        self.__pin = pin
        self.short_duration = short_duration
        self.long_duration = long_duration
        self.short_action = short_action
        self.long_action = long_action
        self.__button = gpiozero.Button(self.__pin, pull_up=True,
                                        hold_time=self.long_duration)
        self.__press_time = None
        self.__button.when_held = self.__on_hold
        self.__button.when_pressed = self.__on_press
        self.__button.when_released = self.__on_release

    def __on_press(self):
        self.__press_time = time.time()

    def __on_release(self):
        release_time = time.time()
        pressed_for = release_time - self.__press_time
        if pressed_for > self.short_duration and pressed_for < self.long_duration:
            # short press
            # do short press stuff here
            try:
                self.short_action()
            except KeyboardInterrupt:
                raise
            except:
                pass

    def __on_hold(self):
        # button held
        # do button held/long press stuff here
            try:
                self.long_action()
            except KeyboardInterrupt:
                raise
            except:
                pass

if __name__ == '__main__':
    def do_short():
        print('short press')

    def do_long():
        print('long press')

    pin = 19
    short_duration = 2
    long_duration =  5

    btn = cmdbutton(pin=pin,
                    short_duration=short_duration,
                    long_duration=long_duration,
                    short_action = do_short,
                    long_action=do_long)
    while True:
        time.sleep(1)
```
All instances work as expected until you stop one of the scripts (Ctrl-C).  Then the other scripts, while still in the True-sleep loop, stop getting events.

```
pi@gpi3:~/button $ ./long_press (first instance)
short press
short press
long press
short press
^CTraceback (most recent call last):
  File "./long_press", line 67, in <module>
    time.sleep(1)
KeyboardInterrupt
pi@gpi3:~/button $
```
```
pi@gpi3:~/button $ ./long_press (second instance)
short press
short press
long press
short press
(Nothing else happens, even tho I keep pressing and releasing the button)
```

lurch, 2019-08-22:
Huh, why are you running the same script multiple times??

When you stop the first script, it closes the GPIO pins https://gpiozero.readthedocs.io/en/stable/faq.html#what-s-the-gpiozero-equivalent-of-gpio-cleanup which presumably puts them into a state in which the other scripts can no longer read from them.

There's also no need for your `while True` loop, see https://gpiozero.readthedocs.io/en/stable/faq.html#keep-your-script-running

And look at https://gpiozero.readthedocs.io/en/stable/faq.html#how-do-i-use-button-when-pressed-and-button-when-held-together if you haven't already :slightly_smiling_face:

tonyyng, 2019-08-23:
> Huh, why are you running the same script multiple times??
The objective isn't to run the identical script multiple times.  It is to use the same button in different scripts that are running at the same time..  It was just easiest to reproduce my issue with a simple script.

Thanks for the links.  I've looked at them, but I don't see a solution that will let the same button be used in multiple scripts at the same time.

lurch, 2019-08-23:
> It is to use the same button in different scripts that are running at the same time.

That's not the way that GpioZero is designed to be used. Can you rearrange your code so that you just have a single script handling the button-presses? What is it you're actually trying to achieve?

tonyyng, 2019-08-23:
> That's not the way that GpioZero is designed to be used. 

It doesn't seem like this is too extreme of a use case (two separate scripts using the same button).  I did look through the docs to see if this was possible.  Is it a reasonable enhancement request?  

> Can you rearrange your code so that you just have a single script handling the button-presses? What is it you're actually trying to achieve?

Do you know if this use case will work if only one script uses gpiozero and other scripts use RPi.GPIO?  Perhaps I'll give that a try.

Another alternative is to have a button monitor script that uses gpiozero and creates /tmp/btn/pressed when the button is pressed and have other scripts look for the file as an indicator that the button is pressed.  Not ideal, but might suit my purposes

My project is a headless PI, where I like to use a [button to have it shutdown cleanly](https://www.raspberrypi.org/magpi/off-switch-raspberry-pi/) with a longer press/hold.  However, I'd like to be able to use short presses of the button to do things in other scripts.

Perhaps [this library](http://abyz.me.uk/rpi/pigpio/python.html) is a better option with a daemon architecture.

I'll report back on the option I end up with.

lurch, 2019-08-23:
> Do you know if this use case will work if only one script uses gpiozero and other scripts use RPi.GPIO? 

I don't think so, as gpiozero uses RPi.GPIO under the hood.

I still think for your use case you'd be better off with a single script - why not have the script that does the shutdown also do the "other things" as well?

If you really want to keep your scripts separate, then I think your best choices would using two separate buttons (one for each script), or using pigpio (as you suggested). @joan2937 - is this a use-case that pigpio supports?

joan2937, 2019-08-23:
@lurch The pigpio daemon (arbitrarily) allows a maximum of 32 concurrent Python scripts.  They can manipulate the same GPIO.  pigpio has no concept of GPIO ownership.  There will be no problems with multiple reads of the same GPIO.

tonyyng, 2019-08-23:
If I use RPi.GPIO this way, it doesn't interfere with the other script using gpiozero:
```#!/usr/bin/env python3

import RPi.GPIO as GPIO
import time

PIN=19
GPIO.setmode(GPIO.BCM)

GPIO.setup(PIN, GPIO.IN, pull_up_down=GPIO.PUD_UP)

while True:
    input_state = GPIO.input(PIN)
    if input_state == False:
        print('Button Pressed')
        time.sleep(0.2)
```
f I use RPi.GPIO with an interrupt and cleanup, the other gpiozero script stops getting events.
```
#!/usr/bin/env python3

import RPi.GPIO as GPIO
import time
import sys
from signal import pause

def buttonEvent(pin):
    print("Pressed")
    GPIO.cleanup()
    sys.exit(0)

PIN=19
GPIO.setmode(GPIO.BCM)

GPIO.setup(PIN, GPIO.IN, pull_up_down=GPIO.PUD_UP)
GPIO.add_event_detect(PIN, GPIO.FALLING)
GPIO.add_event_callback(PIN, buttonEvent)

pause()
```
Given as you point out that gpiozero is using RPi.GPIO under the covers, this makes sense.  I can stick with the non-interrupt RPi.GPIO approach or use pigpio if I need the additional flexibility.

closed 2021-03-12 by bennuttall