gpiozero/gpiozero #795: Set pulse_width on Servo / PWMOutputDevice

bennuttall, 2019-09-17:
It's currently not possible to set the pulse width of a servo or PWM device. There's a `pulse_width` property in `Servo`, but only a getter, not a setter. I added one and tested it with @evildmp and it seemed to work:

```python
@property
def pulse_width(self):
    if self.pwm_device.pin.frequency is None:
        return None
    else:
        return self.pwm_device.pin.state * self.frame_width

@pulse_width.setter
def pulse_width(self, value):
    self.pwm_device.pin.state = value / self.frame_width
```

I think this could be added lower down, in `PWMOutputDevice`.

joan2937, 2019-09-17:
Setting the pulse width is the only correct way of controlling a servo.

A servo moves to a central position if given a 1500µs pulse.  A duty cycle of 7.5% happens to give a 1500µs pulse if PWM is 50Hz.  Change the duty cycle (many servos can cope with lower or higher update rates) and the pulse width also changes which drives the servo to the wrong position,

bennuttall, 2019-09-17:
@joan2937 currently the user can set a "value" between -1 (min pulse width) and 1 (max pulse width). We are calculating the appropriate pulse width and setting it directly, but the user can't set it themselves.

lurch, 2019-09-17:
Given the "abstraction layer" that GpioZero places over servos, what benefit does setting the pulse width directly (instead of just setting its "value") have?

bennuttall, 2019-09-17:
@evildmp has a project where he's calculating the required pulse width and setting directly. He's not even using a linear angle -> pulse width calculation, because he's found it to be inaccurate. He's not currently able to use gpiozero, but is having great success with pigpio.

closed 2019-09-17 by bennuttall

lurch, 2019-09-17:
Ahhh, yeah I guess non-linear servos is a good use-case :slightly_smiling_face:

joan2937, 2019-09-18:
@bennuttall  Is that https://github.com/evildmp/BrachioGraph ?

I like the drawings.  I don't see why it can't be done with gpiozero though.  I assume that there is a mapping from 0..1 to a pulse width and vice versa.

commit 22f068f5 (2019-05-24): Merge pull request #769 from RPi-Distro/fix_overvolt

Fix #795 for overvolted Pis. Also minor tweaks to ASCII-art.
commit 24f679b8 (2019-05-24): Fix #795 for overvolted Pis. Also minor tweaks to ASCII-art.
PR #797 (2019-09-17): Add pulse_width setter to Servo class
Partially fixes #795 - adds pulse_width setter to Servo class. Could we add a getter/setter to `PWMOutputDevice` too? It's not immediately obvious to me how.

@evildmp