gpiozero/gpiozero #799: Python version deprecation policy

bennuttall, 2019-09-19:
I think we should drop support for Python versions when they become EOL.

We currently test on Python 2.7, 3.2, 3.3, 3.4, 3.5, 3.6 and 3.7. Two of these (3.2 and 3.3) are EOL, and another (2.7) is approaching EOL:

Python version | Debian version | EOL | Debian version EOL
--- | --- | --- | ---
2.7 | Buster | 2020-01-01 | 2020-06-30
3.2 | Wheezy | 2016-02-20 | 2018-05-31
3.3 | None | 2017-09-29 | N/A
3.4 | Jessie | 2019-03-18 | 2020-06-30
3.5 | Stretch | 2020-09-13 | 2022-06-??
3.6 | None | 2021-12-23 | N/A
3.7 | Buster | 2023-06-27 | ?

## 3.2 / 3.3

I propose we drop 3.2 and 3.3 from tox config now. They're EOL, must be hardly used, and occasionally cause incompatibility issues we have to spend more time fixing.

## 2.7

I propose that we drop 2.7 from tox config now, and start allowing Python 3-only syntax in the library. Any future releases will support Python 3 only, even if they take place in 2019.

A more conservative approach could be to wait for 1 January to drop it. I just don't see the point creating new Python 2 packages at this late stage. Debian has stopped accepting new Python 2 packages, and is currently removing Python 2 packages from Bullseye (testing).

When we do drop it, there's no need to do a massive refactoring to take advantage of the opportunity for Python only 3 code, but any cleanup can be done as we go along, and any new code can benefit from not having to support both 2 and 3.

## 3.4

Python 3.4 is EOL but its Debian version is not, until June 2020.

I propose we drop 3.4 in time for the Debian Jessie EOL in June.

## 3.5

Python 3.5's EOL is approaching in 2020, but it's in Debian Stretch which is EOL in 2022, so I propose we keep supporting it until then.

## Voting

I propose that the three core maintainers (@waveform80, @lurch and I) should vote on each proposal, and we'll go with the majority for each one.

- Drop Python 3.2 now [y/n]
- Drop Python 3.3 now [y/n]
- Drop Python 2.7 now [y/n]
- Drop Python 2.7 in January [y/n]
- Drop Python 3.4 in June [y/n]

There's no need to vote on newer Python versions for now, but the decisions taken here will provide precedent for future decisions.

bennuttall, 2019-09-19:
1. Drop Python 3.2 now [y]
2. Drop Python 3.3 now [y]
3. Drop Python 2.7 now [y]
4. Drop Python 2.7 in January (if 3 fails) [y]
5. Drop Python 3.4 in June [y]

lurch, 2019-09-19:
I have no strong preferences either way, so I'm happy to go along with whatever Ben & Dave decide.

Another python project I'm involved with also made a similar decision recently https://github.com/PyFilesystem/pyfilesystem2/issues/317

@bennuttall Do you know if all the RPF learning resources / Picademy / etc. have all been updated to be exclusively Python3 ?

bennuttall, 2019-09-20:
> I have no strong preferences either way, so I'm happy to go along with whatever Ben & Dave decide.

In that case, I'll consider your votes to match Dave's.

> Do you know if all the RPF learning resources / Picademy / etc. have all been updated to be exclusively Python3 ?

As far as I'm aware, yes. Picademy always has been, learning resources generally so, and we've removed IDLE2 from the Raspbian image in favour of Thonny and Mu which are Python 3 only.

lurch, 2019-11-17:
Just adding a note here that when this changes, you'll need to remember to update https://gpiozero.readthedocs.io/en/stable/contributing.html#python-2-3 too.

bennuttall, 2020-02-03:
@waveform80 happy to defer some of these calls but can we agree to drop some of them?

i.e. 3.2/3.3 now, 3.4 in June. 2.7 more contentious - I'd rather drop it now but what do you think now January has passed?

If we do a minor release soonish, that could be the last that supports these versions?

lurch, 2020-02-04:
I guess _after_ we drop Python2.7, @XECDesign also needs to stop building python2 deb packages of gpiozero for Raspbian?

> If we do a minor release soonish, that could be the last that supports these versions?

Sounds good to me :+1:

bennuttall, 2020-02-04:
Serge doesn't build them, does he? My understanding is that @waveform80 does and publishes to the repo.

XECDesign, 2020-02-04:
Since it's in Debian now, I believe the plan is to take the package from there in the future.

lurch, 2020-02-04:
Sorry, my mistake! :slightly_frowning_face:

bennuttall, 2020-02-04:
But that's always going to be an older version (it's already quite out-of-date - and that version won't work on Pi 4).

XECDesign, 2020-02-04:
I think the idea is that we'd use their packaging an pull in the latest 'upstream' gpiozero.

bennuttall, 2021-03-14:
Decision: 1.6.0 will be the last to support Python 2. After this we'll support Python 3.5+.

PR #896 (2021-03-18): Drop Python 2.7, 3.2, 3.3, 3.4 #799
closes #799
closed 2021-03-14 by bennuttall