gpiozero/gpiozero #803: close() does not restore original pin state

wallarug, 2019-09-22:
Operating system: 4.19.66-v7+ #1253 SMP Thu Aug 15 11:49:46 BST 2019 armv7l GNU/Linux
Python version: 3.5.3
Pi model: e.g. Pi 3 Model B+
GPIO Zero version: latest
Pin factory used: Default

Hi!

It appears that close() does not set the GPIO pins back to their original states correctly.  After using the gpiozero library the I2C Pins become unusable because the special function has been removed.

Please see below screenshots as evidence with code sample.

**Before**
[image]

**After**
[image]

Is there anyway to set the special functions back after running gpiozero?

```python
from gpiozero import LED, Pin, Button
from time import sleep
import subprocess

print("starting")

def terminal(command):
    MyOut = subprocess.Popen(
      command.split(" "),
      stdout=subprocess.PIPE,
      stderr=subprocess.STDOUT
    )
    stdout, stderr = MyOut.communicate(timeout=60)
    results = [stdout, stderr]
    return results

sda = Button(2)
scl = Button(3)

while True:
    print("running")
    r = terminal("i2cdetect -y 1")
    print(r[0])
    sleep(2)
    sda.close()
    scl.close()

```

bennuttall, 2019-09-22:
@topshed brought this to my attention recently. I think we could probably reset the pin function as well as state on `close()`. I'll look into it. /cc @waveform80

wallarug, 2019-09-22:
Thanks @bennuttall .   I spent a while trying to work out what was going on.  Would be good to restore the state completely.

lurch, 2019-09-23:
...although I guess that would require the pin-library code to read the old "state" (value, mode, function, etc.) of the pin, before modifying it, so that it could restore the 'original' state of the pin when closing it? 

See also the discussion in #323 and #707

wallarug, 2019-09-23:
Technically the cleanup function should put the pin back to its original state anyway as a matter of good practice and by definition.

If you can point me in the direction of where it saves the states currently, maybe I can have a look at fixing this.

lurch, 2019-09-23:
> If you can point me in the direction of where it saves the states currently...

It doesn't; that's part of the problem :wink: 
I simply resets any pin back to a "default state" upon closing it.

wallarug, 2019-09-23:
😟  ok...

Where does it to that?  I had a bit of a dig and there is so much inheritance / OO depth that it is hard to find where the original functions are.

lurch, 2019-09-23:
Indeed :confused:  https://github.com/gpiozero/gpiozero/pull/798#discussion_r327140091

waveform80, 2021-03-19:
>  ok...
> 
> Where does it to that? I had a bit of a dig and there is so much inheritance / OO depth that it is hard to find where the original functions are.

That's down in RPi.GPIO's C code. Basically there's nothing GPIO Zero can do about this; we make a best effort to tell the underlying pin driver to cleanup but it's up to the underlying pin driver (in your case RPi.GPIO) to cleanup properly when we ask it to. In the case of RPi.GPIO, even if we explicitly requested to set the pins back to alt mode 0, we couldn't: its `setup` function only accepts IN or OUT as valid arguments (although the `gpio_function` argument for requesting the current mode returns more values including the I2C and SPI constants). And as noted the `cleanup` function just resets to an input state.

The only suggestion I can offer is that you could try a different underlying pin driver and see whether that resets things differently ,but honestly I doubt it. If I recall correctly none of them store the state of pins on initialization, so there's no way they can restore anything on shutdown.

Arguably, GPIO Zero *could* store the initial state it finds the pin in, and *attempt* to restore that on shutdown (although as noted that would still fail in the RPi.GPIO case above, but it *might* work in the case of other pin drivers). I'm a bit conflicted on that though as in the case of RPi.GPIO (the default pin driver) we'd be left with a choice: silently squash the inevitable error that would occur every single time anyone used the I2C / SPI pins, or print it -- and have everyone using those pins endure a bunch of error messages they can't do anything about every time their script shuts down. I don't *really* fancy either option.

Then there's the whole question of the pin factories supporting remote connections (pigpio, and soon to be rgpio). Should they reset their pins? After all, the daemon controlling the pins is ... still controlling the pins when our script terminates, and may have more than one client anyway (which we can't know about). On the other hand, those factories *know* they're remote. They *could* choose to explicitly do nothing on shutdown.

Or we could add an instance parameter to the factories that could be twiddled by the script, "reset_on_shutdown" on something like that. Default it to true for local factories, and false for remote ones.

@bennuttall any preferences?

waveform80, 2021-03-19:
Hmm, actually - going through the list of issues I've just run across #707 as well. I think in 2.x I need to sit down and have a serious think about what we should do on pin factory shutdown. For now I'm going to close this, but only in favour of having one ticket to track all this - I'll leave an explicit reference in #707 (which has a load more discussion on pin factory shutdown topics too) so we don't forget this.

closed 2021-03-19 by waveform80

wallarug, 2021-03-20:
Thanks @waveform80 .  I'll follow the other issue then.