gpiozero/gpiozero #805: Unable to continue defined function using wait_for_press

kbeflo, 2019-09-27:
Operating system: Raspbian Buster
Python version: 3.7.3
Pi model: Pi 3 Model B
GPIO Zero version: 1.5.1
Pin factory used: e.g. RPiGPIO

From pushing button2 to run a function I was looking to pause and then continue on wait_for_press() timeout or on button press.

```python
from gpiozero import Button

button2 = Button(2)
button3 = Button(3)

def say_hello():
    print("Hello!")

    button3.wait_for_press()
    print("Goodbye!")

button2.when_pressed = say_hello
```

lurch, 2019-09-27:
Urgh, nesting button-handlers like that looks a bit messy. Would something like this also do what you want? (untested)

```python3
from gpiozero import Button
from signal import pause

button2 = Button(2)
button3 = Button(3)

def say_hello():
    print("Hello!")

def say_goodbye():
    if button2.is_pressed:
        print("Goodbye!")

button2.when_pressed = say_hello
button3.when_pressed = say_goodbye

pause()
```

Or maybe you want something like:
```python3
from gpiozero import Button
from time import sleep

button2 = Button(2)
button3 = Button(3)

activated = False

def say_hello():
    global activated
    print("Hello!")
    activated = True

def say_goodbye():
    global activated
    if activated:
        print("Goodbye!")
        activated = False

button2.when_pressed = say_hello
button3.when_pressed = say_goodbye

while True:
    if activated:
        print("I'm active")
    sleep(0.1)
```

kbeflo, 2019-09-30:
I used the first one, thank you for the workaround @lurch. Still looking for a way to use it with a timeout and interrupt it at the same time. 

I still wanted to use button.wait_for_press(). What causes the nested button-handlers to not respond while running the script?

lurch, 2019-10-02:
> Still looking for a way to use it with a timeout and interrupt it at the same time.

Could you give a bit more detail about what exactly it is you're trying to do?

> What causes the nested button-handlers to not respond while running the script?

I suspect that's a question that only @waveform80 will be able to answer :wink: 

All I do know is that event-handlers are (typically) supposed to be very quick-running functions, with any "long-running" logic (which would include any timeouts) being handled inside the main loop, as I illustrated in my second example above.

kbeflo, 2019-10-03:
> Could you give a bit more detail about what exactly it is you're trying to do?

A token generator that displays it on a character LCD on button press. After generating the token I wanted it to timeout after x seconds or on button press then clear the screen and then display "Press button to generate token". Token can only be used once.

For now I just did everything without a timeout.

lurch, 2019-10-03:
Maybe something like this would work?
```python
from gpiozero import Button
from time import sleep, time

button2 = Button(2)
button3 = Button(3)

start_time = None
timeout = 3

def say_hello():
    global start_time
    print("Hello!")
    # show token
    start_time = time()

def say_goodbye():
    global start_time
    if start_time is not None:
        print("Goodbye!")
        # clear screen
        start_time = None

button2.when_pressed = say_hello
button3.when_pressed = say_goodbye

while True:
    if start_time is not None:
        if time() - start_time > timeout:
            say_goodbye()
    sleep(0.1)
```
(as before, this is just written off the top of my head and is entirely untested)

bennuttall, 2019-10-03:
I would use a custom button attribute instead of a global, as per: https://gpiozero.readthedocs.io/en/stable/faq.html#why-do-i-get-an-attributeerror-trying-to-set-attributes-on-a-device-object

closed 2019-10-17 by kbeflo

kbeflo, 2020-04-29:
A bit late but here's the project @bennuttall @lurch https://github.com/kbeflo/unipi. Thank you guys.