gpiozero/gpiozero #809: Nested Parameter Assignment Fails for LEDBoard

briankb, 2019-10-26:
Operating system: Raspbian Buster
Python version: 3.7.3
Pi model: Pi 3 Model B Plus Rev 1.3
GPIO Zero version: 1.5.1
Pin factory used: . RPiGPIO

I'm still a noob in Python so maybe this is simply me misunderstanding how to use the language. I need one of the LED's in the LEDBoard to have the opposite to the default values for active_high and initital_value that are setup for rest of the LEDs. 

Specifically GPIO 6 or as I've named it "Stop" and I've tried two ways to assign the active_high and initial_value parameters.

```python
statusLED = LEDBoard( \
    Blue_Ready=25,Red_Ready=5 \
    ,Blue_Ready_Button=4,Red_Ready_Button=13 \
    ,Start=16,Pause=20,Reset=21 \
    ,InMatch=19,Stop=LEDBoard(StopLED=6, active_high=True, initial_value=False) \
    ,Standby=26,Wait=12, active_high=False, initial_value=True)

for led in statusLED:
    print(led)
    led.on()
    sleep(0.2)
    led.off()
```

the above works until the for loop gets to Stop (GPIO 6) and it throws an exception
```
<gpiozero.LED object on pin GPIO25, active_high=False, is_active=False>
<gpiozero.LED object on pin GPIO4, active_high=False, is_active=False>
<gpiozero.LED object on pin GPIO19, active_high=False, is_active=False>
<gpiozero.LED object on pin GPIO20, active_high=False, is_active=False>
<gpiozero.LED object on pin GPIO5, active_high=False, is_active=False>
<gpiozero.LED object on pin GPIO13, active_high=False, is_active=False>
<gpiozero.LED object on pin GPIO21, active_high=False, is_active=False>
<gpiozero.LED object on pin GPIO26, active_high=False, is_active=False>
<gpiozero.LED object on pin GPIO16, active_high=False, is_active=False>
<gpiozero.LEDBoard object containing 1 devices: redStop and 0 unnamed>
<gpiozero.LED object on pin GPIO12, active_high=False, is_active=False>
Exception in thread Thread-13:
Traceback (most recent call last):
  File "/usr/lib/python3.7/threading.py", line 917, in _bootstrap_inner
    self.run()
  File "/usr/lib/python3.7/threading.py", line 865, in run
    self._target(*self._args, **self._kwargs)
  File "/usr/lib/python3/dist-packages/gpiozero/mixins.py", line 114, in _copy_values
    self.value = v
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 432, in __setattr__
    return super(CompositeDevice, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/devices.py", line 148, in __setattr__
    return super(GPIOBase, self).__setattr__(name, value)
  File "/usr/lib/python3/dist-packages/gpiozero/boards.py", line 145, in value
    for device, v in zip(self, value):
TypeError: zip argument #2 must support iteration
```

I also attempted to updated the assignments after the LEDBoard is created with all defaults. 

```python
statusLED = LEDBoard( \
    Blue_Ready=25,Red_Ready=5 \
    ,Blue_Ready_Button=4,Red_Ready_Button=13 \
    ,Start=16,Pause=20,Reset=21 \
    ,InMatch=19,Stop=6 \
    ,Standby=26,Wait=12, active_high=False, initial_value=True)

statusLED.Stop.active_high = True
statusLED.Stop.initial_value = False
```
Which I thought would work as statusLED.Stop.on() and .off() and .blink() all work so I would assume you could also change the properties of those LED's this way. 

Finally I get the same error when simply trying the example code to use nested LEDboard such as:
```python
statusLED = LEDBoard( \
    Blue_Ready=25,Red_Ready=5 \
    ,Blue_Ready_Button=4,Red_Ready_Button=13 \
    ,Start=16,Pause=20,Reset=21 \
    ,InMatch=19,Stop=LEDBoard(redStop=6) \
    ,Standby=26,Wait=12, active_high=False, initial_value=True)
```

lurch, 2019-10-28:
> Finally I get the same error when simply trying the example code to use nested LEDboard

Have a look at https://gpiozero.readthedocs.io/en/stable/_modules/gpiozero/boards.html#SnowPi to see how nested LEDBoards are designed to be used.

I see you have parameters named `Blue_Ready_Button` and `Red_Ready_Button` - if these are actually buttons rather than LEDs, then maybe you need to use `CompositeDevice` rather than `LEDBoard` ?

bennuttall, 2019-10-28:
Your code example works fine for me:

```
In [1]: from gpiozero import *                                                                       

In [2]: from time import sleep                                                                       

In [3]: statusLED = LEDBoard( \ 
   ...:     Blue_Ready=25,Red_Ready=5 \ 
   ...:     ,Blue_Ready_Button=4,Red_Ready_Button=13 \ 
   ...:     ,Start=16,Pause=20,Reset=21 \ 
   ...:     ,InMatch=19,Stop=LEDBoard(StopLED=6, active_high=True, initial_value=False) \ 
   ...:     ,Standby=26,Wait=12, active_high=False, initial_value=True)                              

In [4]: for led in statusLED: 
   ...:     print(led) 
   ...:     led.on() 
   ...:     sleep(0.01) 
   ...:     led.off() 
   ...:                                                                                              
<gpiozero.LED object on pin GPIO25, active_high=False, is_active=True>
<gpiozero.LED object on pin GPIO4, active_high=False, is_active=True>
<gpiozero.LED object on pin GPIO19, active_high=False, is_active=True>
<gpiozero.LED object on pin GPIO20, active_high=False, is_active=True>
<gpiozero.LED object on pin GPIO5, active_high=False, is_active=True>
<gpiozero.LED object on pin GPIO13, active_high=False, is_active=True>
<gpiozero.LED object on pin GPIO21, active_high=False, is_active=True>
<gpiozero.LED object on pin GPIO26, active_high=False, is_active=True>
<gpiozero.LED object on pin GPIO16, active_high=False, is_active=True>
<gpiozero.LEDBoard object containing 1 devices: StopLED and 0 unnamed>
<gpiozero.LED object on pin GPIO12, active_high=False, is_active=True>

In [5]:  
```

Are you sure that's exactly what you tried?

It's quite hard to read your code. You don't need to use `\` to go onto the next line, and you would be better to use lowercase and underscores (snake case) e.g:

```python
statusLED = LEDBoard(
    blue_ready=25,
    red_ready=5,
    blue_ready_button=4,
    red_ready_button=13,
    start=16,
    pause=20,
    reset=21,
    in_match=19,
    stop=LEDBoard(6, active_high=True, initial_value=0),
    standby=26,
    wait=12,
    active_high=False, initial_value=1
)
```

There's also no need to "name" the nested stop LED.

closed 2019-12-01 by bennuttall