gpiozero/gpiozero #810: Allow tailoring transmit times in EnergenieMaster

goverp, 2019-10-26:
Allow the timings in EnergenieMaster.transmit to be increased from defaults of 0.1/0.25 seconds
This could be in a constructor or setter, or more likely via Environment variables

Purpose of change:
The issue is that I, and I've read others, can't reliably turn an Energenie lightswitch On.  Off is OK.  I suspect the receiver's getting too little power in Off mode (even with a dimmable LED lamp) to reliably detect the signal (and yes, I have soldered an antenna lead to the Pi-mote board).  I found that increasing the timings to 0.2./0.1 seconds lets me switch it on.

My python skills are insufficient to do a proper implementation - at the moment I've just hacked the source package.

bennuttall, 2019-10-28:
Sounds reasonable. I'll take a look.

goverp, 2019-10-28:
On Monday, 28 October 2019 09:58:28 GMT Ben Nuttall wrote:
> Sounds reasonable. I'll take a look.

Thanks.  An alternative I considered afterwards was to add the wait time as a 
parameter to the on command, default 0.25s, which would be more flexible.  In 
retrospect, I suspect the 0.1 sec doesn't need tweaking - IIUC it affects only 
the transmitter, and that's got reliable power.