gpiozero/gpiozero #811: Issues with GPIO When Using Adafruit RGB-Matrix Hat Driver

baker12, 2019-10-29:
Operating system: Raspbian Buster
Python version: 2.7.16 and 3.7.3
Pi model: e.g. Pi 3 Model B+
GPIO Zero version: 1.5.1
Pin factory used: RPiGPIO
See http://rpf.io/gpzissue for information on how to find out these details

Please give us as much information about your issue as possible. Write your
code inside code blocks like so: 
Trying to use the Adafruit RGB Matrix hat with some buttons and when I try running the code with the matrix it acts like the buttons are all pressed. However, when the exact same code is ran without the matrix code is runs just fine. Not sure if it's a bug or something else. Will be happy with any advice. 

Thanks!

```python
#!/usr/bin/env python
import time
import sys

from rgbmatrix import RGBMatrix, RGBMatrixOptions
from PIL import Image

from gpiozero import Button
#from time import sleep

blueButton = Button(21)
redButton = Button(20)
upButton = Button(18) #yellowwire number 3
downButton = Button(17) #greenwire number 4
rightButton = Button(27)  #orangewire number 2
leftButton = Button(22) #redwire number 1 

enter = Image.open('enter.ppm')
gone = Image.open('gone.ppm')
jacks = Image.open('jacks.ppm')
lunch = Image.open('lunch.ppm')
meeting = Image.open('meeting.ppm')
recording = Image.open('recording.ppm')

# Configuration for the matrix
options = RGBMatrixOptions()
options.rows = 32
options.cols = 64
options.chain_length = 1
options.parallel = 1
options.hardware_mapping = 'adafruit-hat'  # If you have an Adafruit HAT: 'adafruit-hat'

matrix = RGBMatrix(options = options)

# Make image fit our screen.
enter.thumbnail((matrix.width, matrix.height), Image.ANTIALIAS)
gone.thumbnail((matrix.width, matrix.height), Image.ANTIALIAS)
jacks.thumbnail((matrix.width, matrix.height), Image.ANTIALIAS)
lunch.thumbnail((matrix.width, matrix.height), Image.ANTIALIAS)
meeting.thumbnail((matrix.width, matrix.height), Image.ANTIALIAS)
recording.thumbnail((matrix.width, matrix.height), Image.ANTIALIAS)

def Press_Blue():
    matrix.SetImage(enter.convert('RGB'))
    
def Press_Red():
    matrix.SetImage(gone.convert('RGB'))
    
def Press_Up():
    matrix.SetImage(jacks.convert('RGB'))
    
def Press_Down():
    matrix.SetImage(lunch.convert('RGB'))
    
def Press_Left():
    matrix.SetImage(meeting.convert('RGB'))
    
def Press_Right():
    matrix.SetImage(recording.convert('RGB'))
    
blueButton.when_pressed = Press_Blue
redButton.when_pressed = Press_Red
upButton.when_pressed = Press_Up
downButton.when_pressed = Press_Down
leftButton.when_pressed = Press_Left
rightButton.when_pressed = Press_Right

#matrix.SetImage(image.convert('RGB'))

try:
    print("Press CTRL-C to stop.")
    while True:
        time.sleep(100)
except KeyboardInterrupt:
    sys.exit(0)
```

bennuttall, 2019-10-29:
Please get rid of your PIL code and reduce it to the most basic version that doesn't work as expected.

At a guess I'd say the buttons are probably pull-down (the default is pull-up), so try `Button(pin, pull_up=False)`.

baker12, 2019-10-29:
So it looks like it might be a bug on the rgbmatrix side. I commented out the 
`
matrix = RGBMatrix(options = options)
`
and that fixed the issue. So it looks like the issue is on their side.

closed 2019-10-31 by bennuttall