gpiozero/gpiozero #812: Button.when_activated lambda assignment in loop is broken

awalasz, 2019-10-29:
Operating system: Raspbian GNU/Linux 9 (stretch)
Python 3.7
Pi model: Pi Zero W
GPIO Zero version: 1.5.1
Pin factory used: None

`when_activated` events assigned in loop causes that for each button when it's pressed the same led blinks (the same last assigned lambda is called `lambda: play_audio(car_3)`

code:
```python
from typing import NamedTuple
from signal import pause

from gpiozero import LED, Button

class CarDefinition(NamedTuple):
    led: LED
    btn: Button
    audio_track: int

car_1 = CarDefinition(led=LED(26), btn=Button(21, pull_up=False), audio_track=0)
car_2 = CarDefinition(led=LED(19), btn=Button(20, pull_up=False), audio_track=1)
car_3 = CarDefinition(led=LED(13), btn=Button(16, pull_up=False), audio_track=2)

def play_audio(car_definition: CarDefinition):
    print("play audio track:", car_definition.audio_track)
    car_definition.led.blink(n=1)

for car in [car_1, car_2, car_3]:
    print(id(car), car.led.pin, car.btn.pin)
    car.btn.when_activated = lambda: play_audio(car)

pause()
```
output after pressing three different buttons:
```
pi@raspberrypiq:~ $ python3.7 main.py
3061389760 GPIO26 GPIO21
3060858928 GPIO19 GPIO20
3060860032 GPIO13 GPIO16
play audio track: 2
play audio track: 2
play audio track: 2
```

replacing 
```python
for car in [car_1, car_2, car_3]:
    print(id(car), car.led.pin, car.btn.pin)
    car.btn.when_activated = lambda: play_audio(car)
```
with:
```python
car_1.btn.when_activated = lambda: play_audio(car_1)
car_2.btn.when_activated = lambda: play_audio(car_2)
car_3.btn.when_activated = lambda: play_audio(car_3)
```
fixes the problem and each led blinks as expected.

awalasz, 2019-10-29:
my bad. This works:
```
for car in [car_1, car_2, car_3]:
    print(id(car), car.led.pin, car.btn.pin)
    car.btn.when_activated = lambda car=car: play_audio(car)
```

closed 2019-10-29 by awalasz

lurch, 2019-10-29:
All the fun of lambda functions and the differences between early and late variable-binding :wink: 

It doesn't really matter, but personally I'd change your last line to:
```python3
    car.btn.when_activated = lambda x=car: play_audio(x)
```
to make it a _little_ bit easier to see what's going on.