gpiozero/gpiozero #813: remote GPIO

Bernd54Albrecht, 2019-11-16:
Operating system: Raspbian Buster
Python version: 3.7
Pi model: Pi 3 Model B (GameHAT/waveshare) and Pi 2 B with WLAN Dongle (Pi2Go lite/4tronix)
GPIO Zero version: 1.5.1
Pin factory used:  RPiGPIO

Two bugs observed, when trying to drive the Pi2Go remotely with the GameHAT:
1. remote_factory instantiated as described. Error message, when using the same BCM number locally and remotely. 
**Solution**:  add     local_factory=PiGPIOFactory()  
and e.g.               button = Button(26, pin_factory = local_factory)
Thus the use of pin numbers is explicit and clear

2. Motor and Robot work fine locally, but not in the remote environment!!!
(LED, PWMLED and Servo did!)
I tried hard for a couple of days, replacing the Motor with 2 LEDs. Locally okay: motor.forward() with one LED on, motor.backward() with the other LED on. Remotely: darkness.
Interim Solution: I instantiated each motor as 2 PWMLEDs: one PWMLED with PWM signal, the other off --> motor running smoothly as wanted. And vice versa.

Unfortunately, I was unable to paste the code of the several test programs and the final product.
More Info is possible via Email.
```python
print("code here")
```

lurch, 2019-11-16:
1. Isn't that expected behaviour? :confused: 

2. I think that's been fixed (in `master`) by #792 but AFAIK there's not been a "packaged release" of GpioZero since then?

Bernd54Albrecht, 2019-11-17:
@ lurch:
ad 1.) Negative, GPIO17 in remote_factory should not block GPIO17 locally. Recommendation for the documentation: Always define remote_factory **and** local factory.
ad 2.) Haven't seen #792, because I looked for the buzz word "remote". Waiting for the next release of gpiozero and then test again.

lurch, 2019-11-17:
@Bernd54Albrecht Hmmm, I'd been assuming that you'd set the default factory to the remote factory, and that's why `Button(26)` (without an explicit pin factory) was creating a pin on the remote Pi. But perhaps I was assuming wrong - could you show code examples that do and don't work?

> Waiting for the next release of gpiozero and then test again.

If you don't want to wait, have a look at [these instructions](https://gpiozero.readthedocs.io/en/stable/development.html#development-installation)

Bernd54Albrecht, 2019-11-17:
@ lurch:
I have problems to insert the code directly. I have basically defined a remote_factory as described in the docs. And as long as I used e.g. LED(26, pin_factory = remote_factory) and Button(13) without any addition, everything worked fine. However, when MotorPin (remote) and Button on the controlling GameHAT had the same pin number, I received error messages and therefore introduced local_factory in my code.

Thanks for the link to the "Development installation". But that is well above my skills level.

lurch, 2019-11-17:
> as long as I used e.g. LED(26, pin_factory = remote_factory) and Button(13) without any addition, everything worked fine.

Cool, that's as expected.

> However, when MotorPin (remote) and Button on the controlling GameHAT had the same pin number, I received error messages

Isn't that because of your issue (2) though, rather than issue (1) being a separate problem?
It's hard to diagnose what's going wrong if you're unwilling to paste sample code.

bennuttall, 2019-11-17:
2 is fixed in #792 as @lurch pointed out.

I'm not clear what the issue with 1 is.

> Solution: add local_factory=PiGPIOFactory()

What is this a solution to? And where are you suggesting it is added?

Please add a simple code example of something you expect to work that doesn't.

Bernd54Albrecht, 2019-11-18:
@bennuttall and lurch
I just sent an Email to ben@raspberrypi.org with 2 attachments, because I am not unwilling, but unable to paste the code here on my Windows computer.
My issue #813 is divided in two subjects: 
1. Attached please find error message in text file. This issue is related to the documentation rather than to the code. 
When I experienced the error message, I simply followed the hint on page 48 of the docs 1.5.1.
My recommendation for clear and unambiguous instantiation of objects:
Always define 
remote_factory = PiGPIOFactory(host="192.168.178.78")   # if necessary, change IP address
local_factory = PiGPIOFactory()

2. Motor, Robot and CamJamKitRobot do not work properly in the remote environment
I apologize that I did not see #792. I was looking for the buzz word "remote", before I decided to open a new issue.
And besides: I do not really unterstand #792
For me, this point is not fixed! 
You have identified the problem, you worked on it, you plan to introduce a fix with the next release. Okay.
I tried hard several days to find out, why my wheels were not turning.

closed 2019-12-01 by bennuttall