gpiozero/gpiozero #814: Command pinout not found

fjp, 2019-11-16:
Operating system: e.g. Ubuntu 18.04
Python version: e.g. 3.6.8 and Python 2.7.15+
Pi model: Pi 4 Model B
GPIO Zero version: 1.5.1
Pin factory used: e.g. RPiGPIO
See http://rpf.io/gpzissue for information on how to find out these details

 I installed gpiozero using the following command and realized that the `pinout`  command isn't working. 

```
sudo apt install python-gpiozero
```

From the [FAQ](https://gpiozero.readthedocs.io/en/stable/faq.html#why-do-i-get-command-not-found-when-running-pinout) in the docs I found that I need to install the python3 version of gpiozero, which I did with the following command:

```
sudo apt install python3-gpiozero
```

Unfortunately I still cannot execute the `pinout`  command. All I get is:

``` 
Command 'pinout' not found, did you mean:

  command 'pinot' from deb pinot

Try: sudo apt install <deb name>
```

Using `sudo pip3 install gpiozero`  didn't work either. Is there anything else I could try to get pinout working? Where is the pinout.py file located when I installed the gpiozero library using the mentioned commands. If I can locate it I will try to execute it directly.

fjp, 2019-11-16:
Using `sudo find / -name "pinout.py"` I located the file at: 

``` 
/home/fjp/.local/lib/python2.7/site-packages/gpiozerocli/pinout.py
``` 

However, when I try to execute it with `python pinout.py` I don't get any output back. Instead I used the following commands on the python console:

```python
from pinout import *
pinout = PinoutTool()
pinout()
```

lurch, 2019-11-16:
IIRC the `.deb` packages are only tested on Raspbian, not on any other distros? I'm afraid that doesn't help identify why it's not working on Ubuntu though...
ping @waveform80

bennuttall, 2019-11-17:
The `pinout` command is not provided by the `python-gpiozero` package only `python3-gpiozero`. However, installing with pip should provide it anyway. The only thing I can think of is that it's not in your PATH. For me, `pinout` is located at `/usr/local/bin/pinout`. If you installed with apt, you should be able to see what files were installed with `dpkg -L python3-gpiozero`, and with pip, just look in dist-packages/site-packges. I'd recommend not installing the Python 2 version for [obvious reasons](https://pythonclock.org/) plus the fact having both could cause confusion.

@waveform80 will be able to clarify the state of gpiozero in Ubuntu as I think he likely packaged it there.

bennuttall, 2019-11-18:
1. Ubuntu 18.04 isn't supported on Pi 4 - how did you install it?
2. The version of gpiozero in 18.04 is 1.3.1, and 1.4.1 in 19.10 - you can't apt install 1.5

There's no reason `sudo pip3 install gpiozero` wouldn't work, provided it doesn't clash with a version provided by apt. It could be that `pip` didn't install anything due to the requirement already being met (an older version).

I'd try uninstalling all gpiozero versions, and install it from scratch using pip3. Would be interested to know how you installed Ubuntu 18.04 on a Pi 4 too.

fjp, 2019-11-18:
Hi @bennuttall and thank you for the support. 

I don't know how this happened but today I rebooted the RPi and now `pinout`  seems to be in my path at `/home/fjp/.local/bin`. Here is the output of `echo $PATH`:

``` 
/opt/ros/melodic/bin:/home/fjp/.local/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/snap/bin
``` 

This means that the command `pinout` is now working. Maybe it is required to do a reboot after installing the package?

To answer your questions:

The reason I used python 2 is because of ROS melodic that I want to run on the RPi 4 B. I installed ubuntu 18.04 on the Pi following [James A Chambers blog post](https://jamesachambers.com/raspberry-pi-4-ubuntu-server-desktop-18-04-3-image-unofficial/). His unofficial releases can be found [here](https://github.com/TheRemote/Ubuntu-Server-raspi4-unofficial/releases) on github. 

For me, `fjp@ubuntu:/usr/local/bin$ ls`  is empty. The output of `dpkg -L python3-gpiozero` is:

``` 
fjp@ubuntu:/usr/local/bin$ dpkg -L python3-gpiozero
/.
/usr
/usr/lib
/usr/lib/python3
/usr/lib/python3/dist-packages
/usr/lib/python3/dist-packages/gpiozero
/usr/lib/python3/dist-packages/gpiozero/__init__.py
/usr/lib/python3/dist-packages/gpiozero/boards.py
/usr/lib/python3/dist-packages/gpiozero/compat.py
/usr/lib/python3/dist-packages/gpiozero/devices.py
/usr/lib/python3/dist-packages/gpiozero/exc.py
/usr/lib/python3/dist-packages/gpiozero/input_devices.py
/usr/lib/python3/dist-packages/gpiozero/mixins.py
/usr/lib/python3/dist-packages/gpiozero/other_devices.py
/usr/lib/python3/dist-packages/gpiozero/output_devices.py
/usr/lib/python3/dist-packages/gpiozero/pins
/usr/lib/python3/dist-packages/gpiozero/pins/__init__.py
/usr/lib/python3/dist-packages/gpiozero/pins/data.py
/usr/lib/python3/dist-packages/gpiozero/pins/mock.py
/usr/lib/python3/dist-packages/gpiozero/pins/native.py
/usr/lib/python3/dist-packages/gpiozero/pins/pigpiod.py
/usr/lib/python3/dist-packages/gpiozero/pins/rpigpio.py
/usr/lib/python3/dist-packages/gpiozero/pins/rpio.py
/usr/lib/python3/dist-packages/gpiozero/spi.py
/usr/lib/python3/dist-packages/gpiozero/spi_devices.py
/usr/lib/python3/dist-packages/gpiozero/threads.py
/usr/lib/python3/dist-packages/gpiozero/tools.py
/usr/lib/python3/dist-packages/gpiozero-1.3.1.post1.egg-info
/usr/lib/python3/dist-packages/gpiozero-1.3.1.post1.egg-info/PKG-INFO
/usr/lib/python3/dist-packages/gpiozero-1.3.1.post1.egg-info/dependency_links.txt
/usr/lib/python3/dist-packages/gpiozero-1.3.1.post1.egg-info/entry_points.txt
/usr/lib/python3/dist-packages/gpiozero-1.3.1.post1.egg-info/requires.txt
/usr/lib/python3/dist-packages/gpiozero-1.3.1.post1.egg-info/top_level.txt
/usr/share
/usr/share/doc
/usr/share/doc/python3-gpiozero
/usr/share/doc/python3-gpiozero/changelog.Debian.gz
/usr/share/doc/python3-gpiozero/copyright
/usr/share/lintian
/usr/share/lintian/overrides
/usr/share/lintian/overrides/python3-gpiozero
```

What do you mean with the following?

> and with pip, just look in dist-packages/site-packges.

And could you provide some help on how to uninstall all gpiozero versions? Thank you.

fjp, 2019-11-18:
>     2\. The version of gpiozero in 18.04 is 1.3.1, and 1.4.1 in 19.10 - you can't apt install 1.5

When I issue the commands from the docs I get:

``` 
>>> from pkg_resources import require
>>> require('gpiozero')
[gpiozero 1.5.1 (/home/fjp/.local/lib/python2.7/site-packages), colorzero 1.1 (/home/fjp/.local/lib/python2.7/site-packages)]
``` 

`pip list`  also shows 

``` 
gpiozero (1.5.1)
``` 

The output of `pip3 list` is:

```
gpiozero (1.3.1.post1)
```

fjp, 2019-11-18:
@bennuttall could you suggest how to proceed if I need Python 2 because of ROS? I guess installing both versions of gpiozero (python2 and 3) like I did will cause conflicts when I try to use gpiozero in some python modules. How can I just use the python2 version of gpiozero and still benefit from your handy pinout command?

bennuttall, 2019-11-18:
If you need to use Python 2, and you're well aware of its deprecation, that's fine. Install with `pip` not `apt` or `pip3`. If you're installing as the user, you'll just need to make sure that where it installs to is in your PATH. Looking at your `require` output, `pinout` would probably be in `/home/fjp/.local/bin`.

fjp, 2019-11-18:
Thank you for the instructions. Would you suggest reinstalling ubuntu 18.04 and gpiozero (using pip) or is there a clean way to remove gpiozero (especially the python3 version)?

And out of curiosity, do you know why the `pinout` command did work after a reboot but not directly after installing gpiozero with `sudo pip3 install gpiozero`?

bennuttall, 2019-11-18:
Not sure. It possibly changed your PATH. Maybe just closing the terminal and starting a new session would have fixed it too.

`sudo apt purge python3-gpiozero` and `sudo pip3 uninstall gpiozero` should do the trick.

closed 2019-11-18 by fjp