gpiozero/gpiozero #815: exceptions handling are not clear

navehgrofi, 2019-12-01:
Operating system: Raspbian stretch
Python version:
Pi model: e.g. Pi 3 Model B+
gpiozero version: 1.5.0
Pin factory used: RPiGPIO
See http://rpf.io/gpzissue for information on how to find out these details

Hi

when im raising exceptions by myself, i get multiple trackback prints even if i try closing all devices before:

Traceback (most recent call last):
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/rpigpio.py", line 244, in _call_when_changed
    super(RPiGPIOPin, self)._call_when_changed()
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/local.py", line 141, in _call_when_changed
    self.state if state is None else state)
  File "/usr/lib/python2.7/dist-packages/gpiozero/pins/pi.py", line 293, in _call_when_changed
    method(ticks, state)
  File "/usr/lib/python2.7/dist-packages/gpiozero/input_devices.py", line 197, in _pin_changed
    self._fire_events(ticks, bool(self._state_to_value(state)))
  File "/usr/lib/python2.7/dist-packages/gpiozero/mixins.py", line 368, in _fire_events
    self._fire_activated()
  File "/usr/lib/python2.7/dist-packages/gpiozero/mixins.py", line 397, in _fire_activated
    super(HoldMixin, self)._fire_activated()
  File "/usr/lib/python2.7/dist-packages/gpiozero/mixins.py", line 344, in _fire_activated
    self.when_activated()
  File "/usr/lib/python2.7/dist-packages/gpiozero/mixins.py", line 334, in wrapper
    return fn(self)
  File "/home/pi/Projects/Mango_Dev/host_ref_code.py", line 175, in interrupt_callback
    UseCases.check_fw_num_of_errs(DBMD4, DBMD2)
  File "/home/pi/Projects/Mango_Dev/UseCases.py", line 2587, in check_fw_num_of_errs
    raise ValueError
ValueError

bennuttall, 2019-12-01:
Can you post your code, and please use backticks to fence your code.

navehgrofi, 2019-12-01:
> Can you post your code, and please use backticks to fence your code.

@bennuttall 

thanks for the reply, the code is actually quite complicated and involves about 6 different files...
can you point me to the possible cause of this? so then i can show the relevant code snippet?

bennuttall, 2019-12-01:
Looks like it's because `check_fw_num_of_errs` takes two mandatory parameters. gpiozero callbacks must take no mandatory parameters (unless it takes one that expects the device object passed in).

Simple example to reproduce your case:

```python
def foo(bar, baz):
    print(bar + " " + baz)

btn.when_pressed = foo
```

Let's say you wanted `foo` to print "hello world" when you pressed the button. You could do:

```python
btn.when_pressed = lambda: foo("hello", "world")
```

or:

```python
def say_hello_world():
    foo("hello", "world")

btn.when_pressed = say_hello_world
```

navehgrofi, 2019-12-10:
@bennuttall 
its strange but its not what happening in my code...
in my code i have the following:
`INT_TRIGGER = gpio_btn(RPI_INTERRUPT_TRIGGER_GPIO, pull_up=False)
INT_TRIGGER.when_pressed = interrupt_callback`

in this case i would expect that when `RPI_INTERRUPT_TRIGGER_GPIO` is going HIGH the `inetrrupt_callback` will be excited (it is happening now correctly)
but, what i get is the errors above which im not sure why...

also, regarding `check_fw_num_of_errs`, it is calling a different function thats controls LED,
the function is simple:
`def blink_LED(target, repeat_time, end_mode = False , led_number = None, led_object = None):
	def turn_on():
		if target == "EVB":
			led_object.on()
		elif target == "RPI":
			shell_command("sudo sh -c 'echo 1 > /sys/class/leds/led0/brightness'")
		else:
			write_to_log("Error while blink led - Not valid device !", "ERROR")
	def turn_off():
		
		if target == "EVB":
			led_object.off()
		elif target == "RPI":
			shell_command("sudo sh -c 'echo 0 > /sys/class/leds/led0/brightness'")
		else:
			write_to_log("Error while blink led - Not valid device !", "ERROR")

	count = 0
	try:
		while count < repeat_time:
			turn_on()
			time.sleep(0.1)
			turn_off()
			time.sleep(0.1)
			count = count + 1
		if end_mode:
			turn_on()
		else:
			turn_off()
	except ValueError:
		write_to_log("LED GPIO IS NOT DEFINED")`

i would love to have more info here

bennuttall, 2019-12-10:
Oh - I have a feeling you can't run shell commands from an event callback.

Although you have posted a different function - can we see what's in `check_fw_num_of_errs`? I assume this calls `blink_LED`? But that is a very complicated function. It defines two functions within its own scope, calls shell commands and iincludes a while loop with a sleep. I think you need to move away from the event-driven paradigm, and bring everything back to the procedural paradigm, i.e. your entire state in a main loop. A simple transition would go from:

```python
btn.when_pressed = foo
btn.when_released = bar
```

to:

```
while True:
    if btn.is_pressed:
        foo()
    else:
        bar()
```

navehgrofi, 2019-12-10:
@bennuttall 

i see, thanks!
now it might make more sense to me, although handling interrupts in this case would be harder as i would need to multi threads my code here.

i will find a way to simplify it

lurch, 2019-12-10:
To expand slightly on what Ben wrote above: common practice is to keep your interrupt-handling code as quick and simple as possible, otherwise there's a risk that you might "miss" the second interrupt, while you're still busy running the code in the first interrupt-handler. A fairly simple way of doing this is to have the interrupt-handlers simply set flags, and then respond to these flags in your main loop.

For example:
```python3
foo_condition = False
bar_condition = False

def set_foo():
    global foo_condition
    foo_condition = True

def set_bar():
    global bar_condition
    bar_condition = True

btn.when_pressed = set_foo
btn.when_released = set_bar

while True:
    if foo_condition:
        process_foo() # this can now be a "slow" function
        foo_condition = False
    if bar_condition:
        process_bar() # this can now be a "slow" function
        bar_condition = False
    sleep(0.1)
```

There are more elegant ways to do this, but using global variables is quick & dirty :wink:  (and it might allow you to avoid writing multi-threaded code)

navehgrofi, 2019-12-11:
thanks @lurch and @bennuttall 

i actually implemented the quick and dirty and ill see if its working well for the long run.
thanks!

closed 2021-02-27 by bennuttall