gpiozero/gpiozero #816: Trouble modifying pin

Th3Whit3Wolf, 2019-12-01:
## Specs
**Operating System:** Raspbian GNU/Linux 10 (buster)
**Python version:** 3.7
**Pi model:** Pi 4 Model B
**GPIO Zero version:** 1.5.1
**Pin factory used:** RPiGPIO

I'm using [**dietpi**](https://www.dietpi.com/)

### Expected Behaviour
Turn off my fan when CPU was below 65 degrees celcius

### What happened
```sh
Traceback (most recent call last):
  File "/home/dietpi/.local/lib/python3.7/site-packages/gpiozero/pins/pi.py", line 109, in pin
    pin = self.pins[n]
KeyError: 4

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "./cpu.py", line 5, in <module>
    fan = OutputDevice(4) # This is the pin the positive is on
  File "/home/dietpi/.local/lib/python3.7/site-packages/gpiozero/devices.py", line 124, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/home/dietpi/.local/lib/python3.7/site-packages/gpiozero/output_devices.py", line 93, in __init__
    super(OutputDevice, self).__init__(pin, pin_factory=pin_factory)
  File "/home/dietpi/.local/lib/python3.7/site-packages/gpiozero/mixins.py", line 106, in __init__
    super(SourceMixin, self).__init__(*args, **kwargs)
  File "/home/dietpi/.local/lib/python3.7/site-packages/gpiozero/devices.py", line 521, in __init__
    pin = self.pin_factory.pin(pin)
  File "/home/dietpi/.local/lib/python3.7/site-packages/gpiozero/pins/pi.py", line 111, in pin
    pin = self.pin_class(self, n)
  File "/home/dietpi/.local/lib/python3.7/site-packages/gpiozero/pins/rpigpio.py", line 132, in __init__
    GPIO.setup(self.number, GPIO.IN, self.GPIO_PULL_UPS[self._pull])
RuntimeError: Not running on a RPi!
```

```python
#!/usr/bin/env python3
import time
from gpiozero import OutputDevice, CPUTemperature

fan = OutputDevice(4) # This is the pin the positive is on
cpu = CPUTemperature()

# Loop Forever
# When hot run fan for 30 seconds and check temp
# When not hot check temp ever 5 secs
while True:
    if cpu.temperature > 65:
        print(cpu.temperature)
        fan.on()
        print("It's Hot Yo!")
        time.sleep(30)
    else:
        print(cpu.temperature)
        fan.off()
        print("Temp seems fine to me homie.")
        time.sleep(5)
```

bennuttall, 2019-12-01:
What do you mean by "trouble modifying pin"?

Do you get the same error simply opening the Python shell and creating an output device? e.g:

```bash
pi@raspberrypi:~ $ python
Python 3.7.3 (default, Apr  3 2019, 05:39:12)
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from gpiozero import OutputDevice
>>> d = OutputDevice(4)
```

Do you get the same issue when using RPi.GPIO? e.g:

```bash
pi@raspberrypi:~ $ python
Python 3.7.3 (default, Apr  3 2019, 05:39:12)
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from RPi import GPIO
>>> GPIO.setmode(GPIO.BCM)
>>> GPIO.setup(4, GPIO.OUT)
```

Th3Whit3Wolf, 2019-12-01:
> Do you get the same error simply opening the Python shell and creating an output device? e.g:

#### Output
```py
Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from gpiozero import OutputDevice
>>> d = OutputDevice(4)
Traceback (most recent call last):
  File "/home/dietpi/.local/lib/python3.7/site-packages/gpiozero/pins/pi.py", line 109, in pin
    pin = self.pins[n]
KeyError: 4

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/dietpi/.local/lib/python3.7/site-packages/gpiozero/devices.py", line 124, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "/home/dietpi/.local/lib/python3.7/site-packages/gpiozero/output_devices.py", line 93, in __init__
    super(OutputDevice, self).__init__(pin, pin_factory=pin_factory)
  File "/home/dietpi/.local/lib/python3.7/site-packages/gpiozero/mixins.py", line 106, in __init__
    super(SourceMixin, self).__init__(*args, **kwargs)
  File "/home/dietpi/.local/lib/python3.7/site-packages/gpiozero/devices.py", line 521, in __init__
    pin = self.pin_factory.pin(pin)
  File "/home/dietpi/.local/lib/python3.7/site-packages/gpiozero/pins/pi.py", line 111, in pin
    pin = self.pin_class(self, n)
  File "/home/dietpi/.local/lib/python3.7/site-packages/gpiozero/pins/rpigpio.py", line 132, in __init__
    GPIO.setup(self.number, GPIO.IN, self.GPIO_PULL_UPS[self._pull])
RuntimeError: Not running on a RPi!
```

> Do you get the same issue when using RPi.GPIO? e.g:

#### Output
```py
Python 3.7.3 (default, Apr  3 2019, 05:39:12) 
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from RPi import GPIO
>>> GPIO.setmode(GPIO.BCM)
>>> GPIO.setup(4, GPIO.OUT)
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
RuntimeError: Not running on a RPi!
```

Is there something I should be using instead of OutputDevice?
I using [this](https://www.amazon.com/GeeekPi-Acrylic-Raspberry-Cooling-Heatsinks/dp/B07TXSQ47L?SubscriptionId=AKIAILSHYYTFIVPWUY6Q&tag=duckduckgo-ffab-20&linkCode=xm2&camp=2025&creative=165953&creativeASIN=B07TXSQ47L) case and have the fan connected to GPIO pings 4 and 6 (GND), I'm simply trying to turn the fan on and off based on a temperature threshold.

bennuttall, 2019-12-01:
There's nothing wrong with your code. It's the underlying pin library failing.

The issue is happening in RPi.GPIO not in gpiozero. For some reason, RPi.GPIO thinks you're not running on a Raspberry Pi. It could be that dietpi isn't exposing the GPIO memory the way Raspbian does.

You could try installing pigpio and switching pin factory:

```bash
sudo apt install pigpio python3-pigpio
sudo pigpiod
```

```bash
pi@raspberrypi:~ $ python
Python 3.7.3 (default, Apr  3 2019, 05:39:12)
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pigpio
>>> pi = pigpio.pi()
>>> pi.write(4, True)
```

If that works, you can use pigpio as your gpiozero pin factory:

```bash
pi@raspberrypi:~ $ GPIOZERO_PIN_FACTORY=pigpio python
Python 3.7.3 (default, Apr  3 2019, 05:39:12)
[GCC 8.2.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> from gpiozero import OutputDevice
>>> d = OutputDevice(4)
>>> d.on()
```

and your previous code will work (as long as you set the pin factory, see [here](https://gpiozero.readthedocs.io/en/latest/api_pins.html))

Alternatively, see if anyone else has had the same problem using RPi.GPIO on dietpi, and see if there's a fix for it.

Alternatively, use Raspbian Lite.

closed 2019-12-01 by Th3Whit3Wolf