gpiozero/gpiozero #817: pin_factory parameter has no effect for Motor/Robot/DistanceSensor

hyansuper, 2019-12-10:
gpiozero version: 1.5.1
It all works fine if I set `gpiozero.Device.pin_factory = PiGPIOFactory(host='pi address')`
But if I don't set pin_factory globally, and specify it in constructor of Motor, Robot, or DistanceSensor, I get error.
```
from gpiozero.pins.pigpio import PiGPIOFactory
import gpiozero
factory=PiGPIOFactory(host='192.168.137.19')
led=gpiozero.LED(3,pin_factory=factory)                  # good
motor=gpiozero.Motor(4,14,pin_factory=factory)    # error
```
error message:
```
Warning (from warnings module):
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\gpiozero\devices.py", line 279
    'Falling back from %s: %s' % (name, str(e))))
PinFactoryFallback: Falling back from rpigpio: No module named 'RPi'

Warning (from warnings module):
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\gpiozero\devices.py", line 279
    'Falling back from %s: %s' % (name, str(e))))
PinFactoryFallback: Falling back from rpio: No module named 'RPIO'
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%
Can't connect to pigpio at localhost(8888)

Did you start the pigpio daemon? E.g. sudo pigpiod

Did you specify the correct Pi host/port in the environment
variables PIGPIO_ADDR/PIGPIO_PORT?
E.g. export PIGPIO_ADDR=soft, export PIGPIO_PORT=8888

Did you specify the correct Pi host/port in the
pigpio.pi() function? E.g. pigpio.pi('soft', 8888)
%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%%

Warning (from warnings module):
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\gpiozero\devices.py", line 279
    'Falling back from %s: %s' % (name, str(e))))
PinFactoryFallback: Falling back from pigpio: failed to connect to localhost:8888

Warning (from warnings module):
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\gpiozero\devices.py", line 279
    'Falling back from %s: %s' % (name, str(e))))
PinFactoryFallback: Falling back from native: [Errno 2] No such file or directory: '/proc/cpuinfo'
Traceback (most recent call last):
  File "C:\Users\hyans\smars.py", line 16, in <module>
    motor=gpiozero.Motor(4,14,pin_factory=factory)
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\gpiozero\devices.py", line 124, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\gpiozero\output_devices.py", line 1224, in __init__
    ('forward_device', PinClass(forward)),
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\gpiozero\devices.py", line 124, in __call__
    self = super(GPIOMeta, cls).__call__(*args, **kwargs)
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\gpiozero\output_devices.py", line 414, in __init__
    pin, active_high, initial_value=None, pin_factory=pin_factory
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\gpiozero\output_devices.py", line 93, in __init__
    super(OutputDevice, self).__init__(pin, pin_factory=pin_factory)
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\gpiozero\mixins.py", line 106, in __init__
    super(SourceMixin, self).__init__(*args, **kwargs)
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\gpiozero\devices.py", line 512, in __init__
    super(GPIODevice, self).__init__(**kwargs)
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\gpiozero\devices.py", line 243, in __init__
    Device.pin_factory = Device._default_pin_factory()
  File "C:\Program Files (x86)\Python37-32\lib\site-packages\gpiozero\devices.py", line 280, in _default_pin_factory
    raise BadPinFactory('Unable to load any default pin factory!')
gpiozero.exc.BadPinFactory: Unable to load any default pin factory!
```

lurch, 2019-12-10:
AIUI this is already fixed (for Motor and Robot) by #792 , it just hasn't yet made it into a "released version" of gpiozero.

It looks like DistanceSensor _should_ work with pin_factory though. https://github.com/gpiozero/gpiozero/blob/v1.5.1/gpiozero/input_devices.py#L749
Could you double-check please?

bennuttall, 2019-12-10:
Yes that's right. You can't (currently) set `pin_factory` for a Motor or Robot, but that's been fixed in master, which will be released as 1.5.2 at some point.

You can install gpiozero from github/master or use this wheel I just made (note it's not a release, consider it a beta): https://bennuttall.com/files/gpiozero-1.5.2b0-py2.py3-none-any.whl

Download this wheel and install on your computer:

```bash
pip install gpiozero-1.5.2b0-py2.py3-none-any.whl
```

In this version, your error should disappear.

hyansuper, 2019-12-10:
Thank you for quick reply.
1. I confirm that DistanceSensor has the same error.
2. In version 1.5.1, if I set global pin_factory and also specify it in parameter, the error disappears. does this mean that global setting takes presidance over parameter setting? I think the opposite should be expected behavior
```
from gpiozero.pins.pigpio import PiGPIOFactory
import gpiozero
factory=PiGPIOFactory(host='192.168.1.102')
gpiozero.Device.pin_factory=factory          
motor=gpiozero.Motor(4,14,pin_factory=factory)    # no error
```
3.Now I have gpiozero of version 1.5.2b0 installed on raspberry pi and my computer, but I still have the error for Motor.

bennuttall, 2019-12-10:
Ah, I see the error in DIstanceSensor. The pin_factory param is passed to the trigger device but not the echo. I'll fix that now and send you a new wheel.

bennuttall, 2019-12-10:
New wheel at https://bennuttall.com/files/gpiozero-1.5.2b1-py2.py3-none-any.whl

> In version 1.5.1, if I set global pin_factory and also specify it in parameter, the error disappears. does this mean that global setting takes presidence over parameter setting? I think the opposite should be expected behavior

You are correct that this is not the intended behaviour. The `pin_factory` param should override the default pin factory. However, in Motor, Robot and DistanceSensor (in 1.5.1), there is a bug that means `pin_factory` param is not passed to the sub-devices. In the case of Motor and Robot, this is all pins. In the case of DistanceSensor, it's just the echo pin. I've fixed these devices, and will make sure we add some tests for other devices just in case. It'll all be fixed for the 1.5.2 release.

> Now I have gpiozero of version 1.5.2b0 installed on raspberry pi and my computer, but I still have the error for Motor.

You don't need gpiozero on your Raspberry Pi, if that's not where you are running the code from. You just need it on your computer. Can you check what version is installed? https://gpiozero.readthedocs.io/en/stable/faq.html#how-can-i-tell-what-version-of-gpiozero-i-have-installed

If you install the new wheel above (1.5.2b1) it should show that as the version. If not, run `pip uninstall gpiozero` and try installing from that wheel.

bennuttall, 2019-12-10:
Oh, and I meant to say, there's a better (and more accurate) explanation of how pin factories are (supposed to be) loaded in the "latest" documentation branch - again, something on master that's not released yet. See it here: https://gpiozero.readthedocs.io/en/latest/api_pins.html (particularly the graph towards the top)

hyansuper, 2019-12-10:
with 1.5.2b1 version as `pkg_resources.require('gpiozero')` shows, I still have the error

bennuttall, 2019-12-11:
I'll leave this open until the fix is in master - thanks for reporting it and helping confirm the fix

commit 23be8f75 (2020-01-20): Merge pull request #830 from gpiozero/fix-817

Ensure pin_factory is sent to both echo and trigger devices in DistanceSensor, close #817
closed 2020-01-20 by bennuttall