gpiozero/gpiozero #818: Exception TypeError: 'super() argument 1 must be type, not None' in <object repr() failed> ignored

navehgrofi, 2019-12-10:
Operating system: Raspbian stretch
Python version:
Pi model: e.g. Pi 3 Model B+
gpiozero version: 1.5.0
Pin factory used: RPiGPIO
See http://rpf.io/gpzissue for information on how to find out these details

Hi

some Exception is jumping without an apparent reason:
`Exception TypeError: 'super() argument 1 must be type, not None' in <object repr() failed> ignored`

my code is very simple, just some inits:

`
VT_CORE_RESET = LED(RPI_VT_CORE_RESET_GPIO)
INT_TRIGGER = Button(RPI_INTERRUPT_TRIGGER_GPIO, pull_up=False)
CPLD_RESET = LED(CPLD_RST_GPIO, active_high=False, initial_value=True)
led_green = LED(GREEN_LED_GPIO)
led_blue = LED(BLUE_LED_GPIO)
led_red = LED(RED_LED_GPIO)`

what could be the reason for it?

lurch, 2019-12-10:
Are your variables:
```
RPI_VT_CORE_RESET_GPIO
RPI_INTERRUPT_TRIGGER_GPIO
CPLD_RST_GPIO
GREEN_LED_GPIO
BLUE_LED_GPIO
RED_LED_GPIO
```
definitely all set to valid numbers first?
Also, I assume your code has linebreaks between statements, and everything all appearing on a single line above was a copy & paste error?

Can you narrow down exactly which line of your code is triggering the error? (e.g. by selectively commenting out lines)

navehgrofi, 2019-12-10:
@lurch 

yes, all variables are defined correctly, and they are basically working in a full project sense.
and yes code has line breaks that for some reason are not seen here.
it seems this line is causing the issue:
`CPLD_RESET = LED(CPLD_RST_GPIO, active_high=False, initial_value=True)`

navehgrofi, 2019-12-10:
@lurch 
it seems to be happening after this function execution:
`def Reset(device, state = "NORMAL"):
	if device.chip_type == "D2":
		reset_device = APR_CORE_RESET
	elif device.chip_type == "CPLD":
		reset_device = CPLD_RESET
	else:
		reset_device = VT_CORE_RESET

	state = str(state.upper())
	reset_device.off()
	time.sleep(0.01)
	if  state == "NORMAL":
		reset_device.on()
		Utilities.write_to_log (device.name + " : Full Reset successfully")
	elif  state == "HOLD":
		Utilities.write_to_log (device.name + " : Holding In Reset successfully")
	elif  state == "RELEASE":
		reset_device.on()
		Utilities.write_to_log (device.name + " : Releasing Reset successfully")
	time.sleep(0.06)	#time for PLL to stable
	if device.chip_type == "CPLD":
		time.sleep(0.04)`

function is called with device set to CPLD and default state

bennuttall, 2019-12-10:
> and yes code has line breaks that for some reason are not seen here.

you need to use three backticks around your code, not one.

bennuttall, 2019-12-10:
Can you post the full code, and the full stack trace? It's impossible to see what's going on with what you've given us.

lurch, 2019-12-10:
If you're not able to post the full code, please try creating a cut-down version that demonstrates the problem you're having. With the snippets you've provided so far, it's impossible to tell if this is a problem with gpiozero, or if the problem is occurring somewhere else...

navehgrofi, 2019-12-10:
@lurch 

im trying to break it down since its complicated,
in general i migrated from Rpi.GPIO where i didnt have any issues

bennuttall, 2020-02-29:
Full stack trace please or I'm closing this issue

closed 2021-02-27 by bennuttall