gpiozero/gpiozero #820: Remote GPIO - No module named 'pigpio'

horchi, 2019-12-16:
I try to use Remote GPIO.
The packages "gpiozero pigpio" are installed:
```
root@gate~> pip3 install gpiozero pigpio
Requirement already satisfied: gpiozero in /usr/local/lib/python3.6/dist-packages (1.5.1)
Requirement already satisfied: pigpio in /usr/local/lib/python3.6/dist-packages (1.44)
Requirement already satisfied: colorzero in /usr/local/lib/python3.6/dist-packages (from gpiozero) (1.1)
```
But the Home Assistand can't find them:
```
root@gate~> grep gpio /var/log/home-assistant.log 
2019-12-16 15:30:52 ERROR (MainThread) [homeassistant.components.binary_sensor] Error while setting up platform remote_rpi_gpio
  File "/srv/homeassistant/lib/python3.6/site-packages/homeassistant/components/remote_rpi_gpio/binary_sensor.py", line 52, in setup_platform
  File "/srv/homeassistant/lib/python3.6/site-packages/homeassistant/components/remote_rpi_gpio/__init__.py", line 36, in setup_input
    from gpiozero.pins.pigpio import PiGPIOFactory
  File "/srv/homeassistant/lib/python3.6/site-packages/gpiozero/pins/pigpio.py", line 42, in <module>
    import pigpio
ModuleNotFoundError: No module named 'pigpio'
15:35:18,root@gate~> pip3 install pigpio
Requirement already satisfied: pigpio in /usr/local/lib/python3.6/dist-packages (1.44)
```

When I check it with a small python (3.6) script the package will be found without a problem.
Dis you habe an idea what went wrong?

bennuttall, 2019-12-16:
What OS are you running? Which user is running the Python script?

horchi, 2019-12-16:
I testet the the python script as root, the home assistant is running with the user homeassistant.
The OS is Ubuntu Server 18.04

bennuttall, 2019-12-16:
Perhaps the homeassistant user doesn't have access to site-packages? Does it run in a virtualenv? Perhaps check with the homeassistant team, or their forums?

horchi, 2019-12-16:
the site-packages seems to be readable for the homeassistant user:
```
root@gate {master} /home/homeassistant/.homeassistant> sudo -u homeassistant head /usr/local/lib/python3.6/dist-packages/pigpio.py 
"""
pigpio is a Python module for the Raspberry which talks to
the pigpio daemon to allow control of the general purpose
input outputs (GPIO).

[http://abyz.me.uk/rpi/pigpio/python.html]

*Features*

o the pigpio Python module can run on Windows, Macs, or Linux
```
```
root@gate {master} /home/homeassistant/.homeassistant> sudo -u homeassistant ls -l  /usr/local/lib/python3.6/dist-packages/*.py
-rw-r--r-- 1 root staff  92913 Nov  9  2018 /usr/local/lib/python3.6/dist-packages/astral.py
-rw-r--r-- 1 root staff   9282 Okt 22  2018 /usr/local/lib/python3.6/dist-packages/daemonize.py
-rw-r--r-- 1 root staff 159822 Okt 22  2018 /usr/local/lib/python3.6/dist-packages/feedparser.py
-rw-r--r-- 1 root staff    779 Okt 22  2018 /usr/local/lib/python3.6/dist-packages/idna_ssl.py
-rw-r--r-- 1 root staff 158393 Dez 16 14:20 /usr/local/lib/python3.6/dist-packages/pigpio.py
-rw-r--r-- 1 root staff  32452 Dez 31  2018 /usr/local/lib/python3.6/dist-packages/six.py
-rw-r--r-- 1 root staff   5341 Okt 22  2018 /usr/local/lib/python3.6/dist-packages/sseclient.py
```

horchi, 2019-12-16:
I have forgotten to change in the virtual environment of the home assistant installation before installing the packages:
sudo -u homeassistant -H -s
source /srv/homeassistant/bin/activate

lurch, 2019-12-16:
This can be closed then?

closed 2019-12-16 by horchi