gpiozero/gpiozero #822: is_lit equivalent for LEDBoard?

TimLavelle, 2019-12-31:
**Short description of change:** Add _**is_lit**_ type attribute to LEDBoard class
**Where would this change be made:** LEDBoard

**Purpose of change:**
As I've been reading the docs, I see that in the LED class, there's a _**is_lit**_ attribute which can be used to check conditions, but is there an equivalent for LEDBoard? 

Context is I've got a series of lights in a LEDBoard and I I'd like to be able to check if it is active (lit) so I can control on/off. 

I can see that LED and PWMLED supports _**is_lit**_ 
But I'm using the pulse setting in the LEDBoard LEDs so I can simulate certain effects across the LEDs

But it seems if I want to be able to control the pulse, and check it's state, I need to use LED/PWMLED vs LEDBoard as LEDBoard does not have an equivalent. 

Please provide example usage of gpiozero with your suggested change: 

```python
body_leds = LEDBoard(6, 5, 13, 19, 16, 26, 20, 21, pwm=True)
hull_led = LEDBoard(18, pwm=True)
left_red_thruster = LEDBoard(27, pwm=True)
left_green_thruster = LEDBoard(22, pwm=True)
right_red_thruster = LEDBoard(4, pwm=True)
right_green_thruster = LEDBoard(17, pwm=True)
boosters = LEDBoard(24, 25, pwm=True)

while True:
	if static_button.is_pressed:
		if body_leds.is_lit:
			leds_off()
		else:
			static_display()
```
Thanks

TimLavelle, 2019-12-31:
So... I was looking at your source code for _**is_lit**_ and saw it pretty much sets `is_active = True`, so I gave that a go and apparently **_is_active_** works and can be used, but it's not specifically documented in the LEDBoard doco.

lurch, 2020-01-03:
I've not done any experimentation, but I _suspect_ that using `.is_active` on an LEDBoard _may_ not work as expected?
The problem is that LEDBoard objects can be nested (see e.g. [SnowPi](https://gpiozero.readthedocs.io/en/stable/api_boards.html#snowpi) ) and I guess an `LEDBoard.is_lit` property would need to (recursively) check if any of the child-LEDBoards have any of their LEDs lit?

Diving into the code, it looks like `LEDBoard.is_active` bubbles down to https://github.com/gpiozero/gpiozero/blob/master/gpiozero/devices.py#L504 and as the `.value` for SnowPi is a tuple of tuples, and a tuple always evaluates as "truthy", I _**think**_ this means that `SnowPi.is_active` will always be `True`, even if all of the LEDs are off? :man_shrugging:

bennuttall, 2020-01-20:
@lurch you can just look at `.leds`:

```python
In [1]: from gpiozero import *                                                  

In [2]: sp = SnowPi()                                                           

In [3]: sp.leds                                                                 
Out[3]: 
(<gpiozero.LED object on pin GPIO23, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO24, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO25, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO17, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO18, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO22, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO7, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO8, active_high=True, is_active=False>,
 <gpiozero.LED object on pin GPIO9, active_high=True, is_active=False>)

In [4]: any(led.is_lit for led in sp.leds)                                      
Out[4]: False
```

But that's silly as `.is_lit` is just an alias for `.is_active` in `LED` which already exists for `LEDBoard`. We can add the alias.

commit 5570f7b9 (2020-01-20): Merge pull request #831 from gpiozero/is-lit-822

Add is_lit alias to LEDCollection, close #822
commit 584d08fb (2020-01-20): Add is_lit alias to LEDCollection, close #822
closed 2020-01-20 by bennuttall