gpiozero/gpiozero #823: Invalid Pointer when importing gpiozero

dhhagan, 2020-01-03:
Operating system: e.g. Raspbian Buster
Python version: e.g. 3.6.5
Pi model: e.g. Pi 3
GPIO Zero version: e.g. 1.5.1
Pin factory used: e.g. RPiGPIO

I have just setup a new(well, old) Pi 3 with the newest version of Raspbian Buster and am trying to use gpiozero to control some GPIO pins. However, when I try importing the library, I see the following:

```python
>>> from gpiozero import Device
free(): invalid pointer
Aborted
```

I searched through the `gpiozero` codebase to see where the error was being thrown, and it appears that it may be outside the scope of `gpiozero` - however, I wanted to check to see if anyone has seen something similar.

lurch, 2020-01-04:
How strange.
`free(): invalid pointer` looks like a 'C' error message rather than a 'Python' error message, so maybe it's coming from RPi.GPIO? Although from a quick search I couldn't find anything relevant at https://sourceforge.net/p/raspberry-gpio-python/tickets/  :confused:

bennuttall, 2020-01-04:
Sorry but this definitely looks like your problem and not ours :)

You're using an old out-of-distribution Python version in Raspbian Buster. Why?

gpiozero works out of the box in Raspbian Buster with Python 3.7.

As lurch said, this isn't a gpiozero issue, it's much lower-level.

I would suggest you stick to Python 3.7 from Raspbian, not use your own Python 3.6 installation. gpiozero is a pure Python module, it's not touching system memory or anything, so this error is nothing to do with gpiozero. If you need to use Python 3.6, you have to accept there'll be some pain. It can't only be gpiozero that fails to import properly. Does RPi.GPIO import? pigpio? Anything else?

closed 2020-01-04 by bennuttall