gpiozero/gpiozero #827: callback after DigitalOutputDevice.blink ends

MNeum, 2020-01-11:
Currently there is no indication, when a DigitalOutputDevice stops blinking when n blinks are done or when the thread was stopped by onother command.
In order to build complex blinking sequences or simply to set the value after blinking a callback would be helpful.
This callback should be called with a parameter indicating whether the blinking stops after the cycle was completed or the cycle was terminated by onother command.

bennuttall, 2020-01-15:
When `background=False`, the blink happens in the foreground, so when it's finished (assuming `n` is set), the blink call returns.

```python
print("starting blink")
led.blink(n=10, background=False)
print("blink completed")
```

Does that help?

MNeum, 2020-01-15:
blink happens in the foreground, everything stops for this time. On a warning flash of 3 times 0.2 sec on/off this means during 1.2 sec no interaction is possible.
It is possible to run this in a separate thread and handle the logic by myself. 
But I imagined the problem of reacting on the end of an output sequence to be common enough to justify a library solution. 

I am working on a demo implementation for my needs, which includes synchronized sequences of led and buzzer. I rather put this into a library than into output classes derived from LED and Buzzer.

lurch, 2020-01-16:
> But I imagined the problem of reacting on the end of a blinking sequence to be common enough to justify a library solution.

AFAIK you're the first person to request this... :man_shrugging:

MNeum, 2020-01-30:
I added this feature and related tests but got a problem while testing.

A lot of output tests (most of them unrelated) fail with:
self = <GPIOThread(Thread-98, started daemon 3036674880)>, timeout = 10

    def join(self, timeout=None):
        super(GPIOThread, self).join(timeout)
        if self.is_alive():
            assert timeout is not None
            # timeout can't be None here because if it was, then join()
            # wouldn't return until the thread was dead
            raise ZombieThread(
>               "Thread failed to die within %d seconds" % timeout)
E           gpiozero.exc.ZombieThread: Thread failed to die within 10 seconds

I ran the tests both on a Raspbian system in a VM and a Raspberry Pi 4+
What could I have done wrong?

lurch, 2020-01-31:
Perhaps the thread is still running after 10 seconds because your "callback function" is still running and keeping it alive? :man_shrugging: 

Threading code always tends to be more complicated than you expect :wink: :cry:

MNeum, 2020-02-02:
It was only complicated because I had an episode of bad eye brain coordination (my brain wouldn't register what my eyes clearly saw).
In the process to untangle my brain wires I encountered something strange though:
several tests define a list of expected states with more states than the mock system state list. Since both lists are zipped, any extra info in the longer list is not tested.
It can easily tested by adding `assert len(self.states) >= len(expected_states)` to MockPin.assert_states and assert_states_and_times. After I did this, 10 tests failed.

Is there a reason for this behavior?

MNeum, 2020-02-05:
Since there seems no desire to implement this in the core library, I suppose we can close this

closed 2020-02-05 by MNeum